1technophile/OpenMQTTGateway #132: New protocol help

bkenobi, 2017-10-15:
I have protocols defined for 3 devices that I would like to provide with the hopes it could be included. I'm not sure how to add these to rcswitch or else I'd add them myself and provide the code.  The devices are sensors that transmit to a base station on 433MHz, so inclusion should be ease.  The devices are an Acurite weatherstation, an Acurite temperature sensor, and a Blueline power monitor.  The protocol and decode procedures are already closed and working in arduino project, just not one based on rcswitch.

1technophile, 2017-10-16:
You could publish your codes so as to enable me to take a look and see if it could be integrated

1technophile, 2017-11-01:
@bkenobi some news about that

Landrash, 2017-11-02:
@1technophile From @bkenobi 's github https://github.com/bkenobi/ookDecoder

1technophile, 2019-02-15:
@bkenobi could you try the last release with ZgatewayPilight uncommented, it supports many more protocols compared to rcswitch?

1technophile, 2019-04-15:
Im' closing this issue, if you have feedback don't hesitate to put it here

closed 2019-04-15 by 1technophile