1technophile/OpenMQTTGateway #133: GatewayRFM69 setup

bkenobi, 2017-10-20:
This seems like a silly question, but I am trying to perform my first compile of the code.  For starters, I downloaded the whole package as well as the individual libraries.  I extracted all libraries into the OpenMQTTGateway/libraries path.  It appears that in Arduino 1.8.5 the libraries must be in the install path and not in the sketch path.

I've read through the wiki and don't see any guidance on how to set up each of the configurations.  I want to use the code to create an ESP8266 + RFM69 to MQTT gateway.  I read through the User_config.h and see that I need to comment out the Z... files that I don't need.  I have the user_config.h file set as follows:

```
/*-------------DEFINE THE MODULES YOU WANT BELOW----------------*/
//Addons and module management, comment the Z line and the config file if you don't use
#ifdef ESP8266 // for nodemcu, weemos and esp8266
//  #define ZgatewayRF
//  #include "config_RF.h"
//  #define ZgatewayRF2
//  #define ZgatewayIR
//  #include "config_IR.h"
//  #define ZgatewayBT
//  #include "config_BT.h"
//  #define ZsensorHCSR501
//  #include "config_HCSR501.h"
//  #define ZsensorADC
//  #include "config_ADC.h"
//  #define ZsensorBH1750
//  #include "config_BH1750.h"
//  #define ZsensorBME280
//  #include "config_BME280.h"
  //#define ZsensorDHT // If you uncomment this you can't use I2C due to the fact that I2C use also D1
  //#include "config_DHT.h"
  #define ZgatewayRFM69 // If you uncomment this you can't use RF and BT due to the fact that RF use also D8 and BT use also D6/D7
  #include "config_RFM69.h"
```

I think this is as needed but I get errors during compile.

`OpenMQTTGateway\OpenMQTTGateway.ino: In function 'boolean isAduplicate(long int)':

OpenMQTTGateway:345: error: 'time_avoid_duplicate' was not declared in this scope

       if (now - ReceivedSignal[i][1] < time_avoid_duplicate){ // change

                                        ^

exit status 1
'time_avoid_duplicate' was not declared in this scope`

Perhaps I'm missing something obvious.  I'm really just trying to get this to compile currently.

1technophile, 2017-10-20:
No you are not, i reproduce it, i will publish a correction. By the time you can move the time_avoid_duplicate declaration from config_rf.h to user_config.h.

1technophile, 2017-10-20:
master branch updated, please install it and close the issue if ok for you

bkenobi, 2017-10-25:
I downloaded and compiled.  All is ok.

That said, I'm still not clear on what configuration I should be using.  I read through the wiki and know what the hardware config should be.  I'm just not sure what needs changing in the User_config.h(or other files) to get the correct setup.  I want to use ESP8266 (Wemos D1) + RFM69 to receive 433MHz and then broadcast on MQTT.  I see in the hardware compatibility forum there are MANY people who have had no issue.  I can't compile with the default configuration and slimming down to just ZgatewayRFM69 doesn't seem to work either.  I only changed the #define and #include sections below.  Any suggestions or locations where a guide has been posted would help immensely.

`/*  
  OpenMQTTGateway  - ESP8266 or Arduino program for home automation 
   Act as a wifi or ethernet gateway between your 433mhz/infrared IR signal  and a MQTT broker 
   Send and receiving command by MQTT
 
  This program enables to:
 - receive MQTT data from a topic and send RF 433Mhz signal corresponding to the received MQTT data
 - publish MQTT data to a different topic related to received 433Mhz signal
 - receive MQTT data from a topic and send IR signal corresponding to the received MQTT data
 - publish MQTT data to a different topic related to received IR signal
 - publish MQTT data to a different topic related to BLE devices rssi signal
  
    Copyright: (c)Florian ROBERT
  
    This file is part of OpenMQTTGateway.
    
    OpenMQTTGateway is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.
    OpenMQTTGateway is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.
    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

/*----------------------------USER PARAMETERS-----------------------------*/
#define SERIAL_BAUD   115200
/*-------------DEFINE YOUR NETWORK PARAMETERS BELOW----------------*/
//MQTT Parameters definition
#define mqtt_server "192.168.0.200"
//#define mqtt_user "your_username" // not compulsory only if your broker needs authentication
//#define mqtt_password "your_password" // not compulsory only if your broker needs authentication
#define mqtt_port 1883
#define Gateway_Name "OpenMQTTGateway"
#define will_Topic "home/OpenMQTTGateway/LWT"
#define will_QoS 0
#define will_Retain true
#define will_Message "Offline"
#define Gateway_AnnouncementMsg "Online"

/*-------------DEFINE YOUR NETWORK PARAMETERS BELOW----------------*/
// Update these with values suitable for your network.
#ifdef ESP8266 // for nodemcu, weemos and esp8266
  #define wifi_ssid "wifi ssid"
  #define wifi_password "wifi password"
#else // for arduino + W5100
  const byte mac[] = {  0xDE, 0xED, 0xBA, 0xFE, 0x54, 0x95 }; //W5100 ethernet shield mac adress
#endif

const byte ip[] = { 192, 168, 0, 99 }; //ip adress
// Advanced network config (optional) if you want to use these parameters uncomment line 158, 172 and comment line 171  of OpenMQTTGateway.ino
const byte gateway[] = { 192, 168, 1, 1 }; //ip adress
const byte Dns[] = { 192, 168, 1, 1 }; //ip adress
const byte subnet[] = { 255, 255, 255, 0 }; //ip adress

/*-------------DEFINE YOUR OTA PARAMETERS BELOW----------------*/
#define ota_hostname "OTAHOSTNAME"
#define ota_password "OTAPASSWORD"
#define ota_port 8266

/*-------------DEFINE THE MODULES YOU WANT BELOW----------------*/
//Addons and module management, comment the Z line and the config file if you don't use
#ifdef ESP8266 // for nodemcu, weemos and esp8266
//  #define ZgatewayRF
//  #include "config_RF.h"
//  #define ZgatewayRF2
//  #define ZgatewayIR
//  #include "config_IR.h"
//  #define ZgatewayBT
//  #include "config_BT.h"
//  #define ZsensorHCSR501
//  #include "config_HCSR501.h"
//  #define ZsensorADC
//  #include "config_ADC.h"
//  #define ZsensorBH1750
//  #include "config_BH1750.h"
//  #define ZsensorBME280
//  #include "config_BME280.h"
  //#define ZsensorDHT // If you uncomment this you can't use I2C due to the fact that I2C use also D1
  //#include "config_DHT.h"
  #define ZgatewayRFM69 // If you uncomment this you can't use RF and BT due to the fact that RF use also D8 and BT use also D6/D7
  #include "config_RFM69.h"
#else // for arduino + W5100
  #define ZgatewayRF
  #include "config_RF.h"
  //#define ZgatewayRF2 // too big for UNO
  //#define ZgatewayRFM69 not tested
  //#include "config_RFM69.h"
  #define ZgatewayIR
  #include "config_IR.h"
  #define ZgatewayBT
  #include "config_BT.h"
  //#define ZsensorDHT
  //#include "config_DHT.h"
  //#define ZsensorBH1750
  //#include "config_BH1750.h"
  //#define ZsensorBME280
  //#include "config_BME280.h"
  //#define ZsensorHCSR501
  //#include "config_HCSR501.h"
  //#define ZsensorADC
  //#include "config_ADC.h"
  //#define ZgatewayRFM69 not tested
  //#include "config_RFM69.h"  
#endif
/*----------------------------OTHER PARAMETERS-----------------------------*/
/*-------------------CHANGING THEM IS NOT COMPULSORY-----------------------*/
/*--------------MQTT general topics-----------------*/
// global MQTT subject listened by the gateway to execute commands (send RF, IR or others)
#define subjectMQTTtoX "home/commands/#"
#define subjectMultiGTWKey "toMQTT"

//variables to avoid duplicates
#define time_avoid_duplicate 3000 // if you want to avoid duplicate mqtt message received set this to > 0, the value is the time in milliseconds during which we don't publish duplicates

/*-------------------ACTIVATE TRACES----------------------*/
#define TRACE 1 // 0= trace off 1 = trace on`

1technophile, 2017-10-25:
If compile is ok you now have to define your rfm69 parameters in config_RFM69.h

PS:I suppose that as you want to use RFM69 you have other RFM69 module available for sending and receiving data?

bkenobi, 2017-10-25:
Sorry, I don't think I explained clearly.

I want to use the ESP8266 with RF to receive from 433MHz sensors (weather station, power meter, etc).  I want to use the RFM69 due to superior range.  I am not planning on using multiple OpenMQTTGateways to communicate with each other.  I think if I used a simple 433MHz receiver, I'd configure with ZgatewayRF and ZgatewayRF2 only.  But, in order to use RFM69 I have to enable ZgatewayRFM69.  However, that appears designed for communication between devices.  Maybe I should start off by just installing a 433MHz receiver module first?

1technophile, 2017-10-26:
"Maybe I should start off by just installing a 433MHz receiver"
Yes you should start with 433mhz receiver and emitter like STX882 & SRX882.
Rfm69 is integrated in a way that it can talk only to other rfm69.
ZgatewayRF is compatible with this list of devices : 
https://docs.google.com/spreadsheets/d/1DdtVtSsN25nwP6BZI5q6C9yDGz37tUWjw2SQ1RGwBxU/edit#gid=0

bkenobi, 2017-10-26:
I understand that weather station protocols are not currently incorporated. I know the protocol and how to decode it, so I'm hoping inclusion won't be a problem.

If RFM69 is only included for gateway-to-gateway communication, is there any was to make it work as a standard receiver? It's range is supposed to be far superior over other rf modules. I know it can be done as I have an example for use with Arduinos. I don't know if it's easily accomplished for use with rcswitch.

1technophile, 2017-10-26:
Yes there is a way by using another library instead of the current one. But this needs some research.

flangeslag, 2017-10-27:
Hi, I'm getting a compiling error too:

Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board:"Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

C:\Users\Fred_2\Documents\Arduino\OpenMQTTGateway\OpenMQTTGateway.ino: In function 'void setup()':

OpenMQTTGateway:138: error: 'SERIAL_BAUD' was not declared in this scope

     Serial.begin(SERIAL_BAUD, SERIAL_8N1, SERIAL_TX_ONLY);

                  ^

exit status 1
'SERIAL_BAUD' was not declared in this scope

I would I preciate it if you provide a solution for this for I'm not a programmer, thanks in advance, Fred

1technophile, 2017-10-27:
Hi,

you are trying to set up the RFM69 gateway?

flangeslag, 2017-10-27:
Hi, thank you for the fast response, and Yes, I think it's officially
called "openmqttgateway", I'm trying to set up the "RF" part.

1technophile, 2017-10-27:
Ok this issue is related to rfm69. You should better use ZgatewayRF. If you have questions , support needs please go to the gitter tchat on the readme page.

flangeslag, 2017-10-27:
If you have to setup an environment like mine, don't bother you can use,
mine.
Regards
Fred

1technophile, 2017-10-27:
Sorry i don't understand

flangeslag, 2017-10-27:
Hi , what is it you don't understand, my request on " openMqttGateway" or
using my environment?

1technophile, 2017-11-01:
@bkenobi could we close this issue as it seems a little bit confusing with these different posts. If you want to include code for weather station you can open another issue with this specific need or submit a pull request

flangeslag, 2017-11-02:
Sure we can close this issue
Fred

closed 2017-11-02 by 1technophile