1technophile/OpenMQTTGateway #134: RF only receives when serial console is used (ESP8266)

lademeister, 2017-10-20:
I am using OpenMQTTGateway 0.4.
Great work btw!
When receiving RF Signals (using RXB-8 Superheterodyne) I only Receiver Signals when The ESP8266 is connected to my Laptop AND serial console in Arduino IDE so opebed. When closing  serial console IR and DHT still works but RF receive immediately stops receiving (Prozent publishing to home/433toMQTT.
DHT and IR still publish. 
As soon as I open serial console it starts working again.
On any other power source only IR and DHT work (assumably because there is no serial connection).
What causes this behavior? How could I get it working without serial connection?

BertrumUK, 2017-10-20:
Not sure if this will help but I finally got around to getting my 433 to mqtt working yesterday in Home Assistant. Here is my code if it help

alias: "MQTT RF 3 On"
  initial_state: True
  hide_entity: False
  trigger:
    platform: mqtt
    topic: home/433toMQTT
    payload: 5578557
  action:
    service: homeassistant.turn_on
    entity_id: switch.fireplace

BertrumUK, 2017-10-20:
Sorry about the spacing. Will upload the code tomorrow my site

https://github.com/BertrumUK/home-assistant

BertrumUK, 2017-10-20:
What do you see when monitoring MQTT messages when disconnected from the laptop and serial monitor is removed? Anything via home/433toMQTT ?

lademeister, 2017-10-20:
Actually it's an issue only referring to the OpenMQTTGateway software itself, has nothing to do with implementation in home automation (I am using openHAB and that part works).
Basically my problem is that the ESP8266 on which I am running OpenMQTTGateway 0.4 stops publishing received RF signals in the corresponding MQTT topic as soon as I close serial console on Arduino IDE.
Everything else still works...
That is somehow very strange behavior because it does not make any sense to depend on a serial connection...

lademeister, 2017-10-20:
Nothing is being published to home/433toMQTT as soon as I close the console. But DHT temperature and humidity and as well received IR commands are still being published.
When I open the serial console it starts working again (repeatable)

BertrumUK, 2017-10-20:
Yes strange, Are you running the latest versions ? OpenMQTTGateway.ino should be pulling the 433toMQTT code and sending it to the broker. Is there a chance that any of your code is has "//" before it on the 433toMQTT side so is not being sent to the broker ?

lademeister, 2017-10-20:
No - because it starts sending when I open the serial debug console in Arduono IDE - if O had commented out anything by accident it would not publish at all...

1technophile, 2017-10-20:
Hello,

To which pin your receiver is connected ? Could you try another pin and change it into config_rf.h

lademeister, 2017-10-20:
Hello 1technophiIe!

To be precise I am using a Wemos D1 Mini Board (ESP8266).

Currently RF Receiver is on pin D3, while DHT22-Sensor is on pin D5. IR is not connected (but configured) because it gave an compilation error that said something about IR when commenting the lines 
  #define ZgatewayIR
  #include "config_IR.h"
to 
// #define ZgatewayIR
//  #include "config_IR.h"
in User_config.h so I left it in. 
Also there is no RF sender (but also configured) - I also did not use identical pin definitions (e.g. for IR and RF). 
I double checked that.
I then wanted to be sure there is no mapping error with the "D1-D8" pins on ESP (which always worked, but anyways..) - so I defined:
in config_RF.h:
RF_RECEIVER_PIN 0 
(which is D3 on Wemos D1 Mini)
in config_DHT.h:
DHT_RECEIVER_PIN 14 
(which equals D5 on Wemos D1 Mini)
But I also tried "D3" and "D5", I also tried different pins for DHT, IR and RF and especially I tried interchanging working ones - I know that IR receive is working because I tried with some input.
What I can say by now that IR receive works on all pins (I tried D1-D8) while RF receive doesn't work on any pin. 
I do only know for sure that RF receive starts working with an opened serial console on Pin D3.
I did not check for the use of other pins if RF receive starts working with an opened serial console, but I can check that as well. 
I re-downloaded the OpenMQTTGateway and also RC-switch.
Today I put some debug code in ZGatewayRF.ino and I am now totally positive that the problem is that if(mySwitch.available()) in 
boolean RFtoMQTT(){
will only become true when serial console in Arsuino IDE is opened.

ZGatewayRF.ino:

boolean RFtoMQTT(){
//code gets here always
  if (mySwitch.available()){
//code gets here only when receiving RF data AND an opened serial console 
    trc(F("Rcv. RF"));
[...]
That does not make any sense to me..

1technophile, 2017-10-21:
Are you sure of the ppwer supply voltage of your board ?

lademeister, 2017-10-21:
The board can be powered by USB and it has a power regulator so it provides 5V and 3,3V from that 5V USB.
I tried 1A and also 2A power source (but I think that the laptop provides only 500mA on USB ports and there it works.

1technophile, 2017-10-21:
You re rx and tx pins are they connected to something ?

lademeister, 2017-10-21:
Today I tested if the RX pin for RF SEND causes the problem and changed it to another pin, but that does not change anything.
What I found - and that's even more strange - is that if I use a different Wemos D1 mini board (original Wemos, the other one is a quality rebuilt with 32MB flash) it behaves a little different: RF receive works as soon as the board is being powered by my laptops USB port - even without opening serial console.
But as soon as I power it from an iPhone USB charger it does not publish RF to MQTT. DHT still working regardless of power source

lademeister, 2017-10-21:
Found it. I added pinmode input for the receiver pins because I couldn't get it to work at first. This caused the problem. Sorry for the confusion.

1technophile, 2017-10-21:
ok no problem

lademeister, 2017-10-23:
Works perfectly. I am using it with openHAB. Great piece of software!

closed 2017-10-23 by lademeister