1technophile/OpenMQTTGateway #135: Add ability to view log output via TELNET

cosmocracy, 2017-10-27:
This issue covers the work to integrate JoaoLopesF's excellent [ESP8266-RemoteDebug-Telnet](https://github.com/JoaoLopesF/ESP8266-RemoteDebug-Telnet) library/facility into the gateway.  This would allow for headless debugging/troubleshooting as diagnostic logging could be viewed remotely by a simple Telnet connection to the gateway.

Given the security implications of making internals available externally, the implementation should be configurable so that deployments that do not want/need remote debugging can easily disable the feature.

cosmocracy, 2017-10-27:
I would be interesting in contributing to this issue and have forked the repository.

cosmocracy, 2017-10-28:
The `trc(String msg)` method looks like a natural place to intercept trace logging and send it to  RemoteDebug.

1technophile, 2017-10-28:
Hello, feel free to submit a pull request, it can be interesting indeed, maintenance could be done without wire with that.  Thanks for the idea.

1technophile, 2018-01-06:
@cosmocracy, maybe a simpler way, what do you think if we publish the debug messages on a special mqtt topic  ?

scotthernandez, 2018-01-10:
Yep, mqtt logging seems like a better idea since it is already a requirement, libraries are included, and won't require any more server resources or knowledge from the user (if they need help subscribing to the topic the wiki can include simple directions).

1technophile, 2019-02-15:
done in development

closed 2019-02-15 by 1technophile