1technophile/OpenMQTTGateway #136: Add repeat for IR_PANASONIC

glasruit, 2017-11-02:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

So as to have a more efficient treatment of your request please provide the following information:
* OpenMQTTGateway version used (V0.4, 0.3, master...)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries
* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the tchat room below
[image]

glasruit, 2017-11-02:
My Panasonic tv needs 3 IR repeats to switch on. I managed to do so by altering a few lines of your code.

In config_IR.h line 43 add:
`#define PanasonicBits         48U        // Panasonic nr of bits (Pre data)`

In ZgatewayIR.ino line 250 change
`irsend.sendPanasonic(PanasonicAddress, data);`
into
`irsend.sendPanasonic(PanasonicAddress, data, PanasonicBits, valueRPT);`

TV on with:
`mosquitto_pub -t home/commands/IR_PANASONIC/RPT_3 16825533`

1technophile, 2017-11-02:
Ok i will add it

closed 2017-11-02 by 1technophile

glasruit, 2017-11-03:
Sorry for not being clear at the first instance. Only TV-on needs 3 IR repeats. All other IR commands are single IR actions. Please delete line 251 in  `ZgatewayIR.ino`
`if (valueRPT == 0) valueRPT = 2;`

By the way, thank you for al your code.