1technophile/OpenMQTTGateway #137: help wanted IR sensor

sumitrajancse, 2017-11-05:
I am making an IR sensor with ESP8266 to work with hassio.
Following is my code from the configuration.yaml file
`switch:
  - platform: mqtt
    name: "Fan On/Off"
    state_topic: "home/IRtoMQTT" 
    command_topic: "home/commands/IR_Raw"
    payload_on: "ON"
    payload_off: "OFF"
    optimistic: false
    retain: true`

I upload the code to the nodeMCU as per the description.
No parameters were changed for the state and topic of MQTT.

On the serial monitor , when i press the remote of the FAN, I have the following code for the FAN (Fan On)
`Rcv. IR
104,104,728,104,208,832,104,312,416
--no pub. unknown protocol--

`
Following is the output on the serial when when i use the following command

` mosquitto_pub -t home/commands/IR_Raw  -m  "104,104,728,104,208,832,104,208,520" -h 192.168.1.27
`

Output on the serial 

`Hey I got a callback 
104,104,728,104,208,832,104,208,520
IR_Raw
MQTTtoIR ack pub.
Hey I got a callback 
Storing signal
Min ind: 
1
store code :
104/135189
Col: val/timestamp
0:104/5478
1:104/135189
2:0/0
3:0/0
4:0/0
5:0/0
6:0/0
7:0/0
8:0/0
9:0/0
10:0/0
11:0/0
Data stored
104,104,728,104,208,832,104,208,520
`
Nothing happens when I try to turn on the FAN ..
Can you please help

1technophile, 2017-11-05:
Hello what is the brand of your device ?

sumitrajancse, 2017-11-05:
Its a normal IR fan.. brand Lasko

1technophile, 2017-11-05:
take a look to that after some googling:
https://github.com/z3t0/Arduino-IRremote/issues/528

1technophile, 2017-11-06:
for questions I advise to go to the tchat more than submitting an issue. Please post your investigations there:
https://gitter.im/OpenMQTTGateway/Questions_support

closed 2017-11-06 by 1technophile