1technophile/OpenMQTTGateway #138: Does't work without the serial attached

marci144, 2017-11-11:
It has to be a very stupid problem ....

I use the gateway for the RF433 and everything works ..... as long I keep the serial monitor open or a PUTTY on the COM port. If they are disconnected the gateway stops working.

So I have a bit of a problem deploying in the final location.

I tried to disable TRACE, I have only the message from writeln on the console, but I'm in the same situation... No console, nothing works

Tks in advance for help
Marcello

1technophile, 2017-11-11:
Hello,

Try to connect receiver to d2 (gpio 4) instead of d3 and change the pin in config_rf

marci144, 2017-11-11:
Problem solved ! Many tks 
GPIO0 is always peculiar!
tks again
Marcello

1technophile, 2017-11-11:
Great , could you try to connect it to rx pin and say me if it works ?

marci144, 2017-11-11:
I'm back!

tried on GPIO3 RX and it doesn't work at all. Even with the serial console on nothing happen
BR
Marcello

marci144, 2017-11-11:
gpio3 RX doesn't work at all !
Marcello

closed 2017-11-11 by marci144

1technophile, 2017-11-11:
thx for the feedback