1technophile/OpenMQTTGateway #140: how to get 315 amd 433mhz working together with one OpenMQTTGateway？

jxjhheric, 2017-11-20:
is it possible？

1technophile, 2017-11-20:
I will try first to connect both emitter data pins together. Regarding receiver i think you will have to duplicate the gateway  and connect each receiver to different pins;  produve several instance of myswitch depending on the pin. Do you feel confident on doing that or do you want some draft sketch ?

jxjhheric, 2017-11-20:
If there is some draft sketches, that is the best!

1technophile, 2017-12-07:
Add a file called ZgatewayRF315 with the content below

```
/*  
  OpenMQTTGateway  - ESP8266 or Arduino program for home automation 

   Act as a wifi or ethernet gateway between your 433mhz/infrared IR signal  and a MQTT broker 
   Send and receiving command by MQTT
 
  This gateway enables to:
 - receive MQTT data from a topic and send RF 433Mhz signal corresponding to the received MQTT data
 - publish MQTT data to a different topic related to received 433Mhz signal

    Copyright: (c)Florian ROBERT
  
    This file is part of OpenMQTTGateway.
    
    OpenMQTTGateway is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    OpenMQTTGateway is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/
#ifdef ZgatewayRF315

#include <RCSwitch.h> // library for controling Radio frequency switch

RCSwitch mySwitch315 = RCSwitch();

void setupRF315(){

  //RF init parameters
  mySwitch315.enableTransmit(RF_EMITTER_PIN315);
  trc(F("RF_EMITTER_PIN 315"));
  trc(String(RF_EMITTER_PIN315));
  mySwitch315.setRepeatTransmit(RF_EMITTER_REPEAT315); 
  mySwitch315.enableReceive(RF_RECEIVER_PIN315); 
  trc(F("RF_RECEIVER_PIN 315"));
  trc(String(RF_RECEIVER_PIN315));
}

boolean RF315toMQTT(){

  if (mySwitch315.available()){
    trc(F("Rcv. RF"));
    unsigned long MQTTvalue = 0;
    String MQTTprotocol;
    String MQTTbits;
    String MQTTlength;
    MQTTvalue = mySwitch315.getReceivedValue();
    MQTTprotocol = String(mySwitch315.getReceivedProtocol());
    MQTTbits = String(mySwitch315.getReceivedBitlength());
    MQTTlength = String(mySwitch315.getReceivedDelay());
    mySwitch315.resetAvailable();
    if (!isAduplicate(MQTTvalue) && MQTTvalue!=0) {// conditions to avoid duplications of RF -->MQTT
        trc(F("Adv data RFtoMQTT"));
        client.publish(subjectRF315toMQTTprotocol,(char *)MQTTprotocol.c_str());
        client.publish(subjectRF315toMQTTbits,(char *)MQTTbits.c_str());    
        client.publish(subjectRF315toMQTTlength,(char *)MQTTlength.c_str());    
        trc(F("Sending RF315toMQTT"));
        String value = String(MQTTvalue);
        trc(value);
        boolean result = client.publish(subjectRF315toMQTT,(char *)value.c_str());
[... 137 lines cut ...]
    #define RF315_RECEIVER_PIN 1 //1 = D3 on arduino
    #define RF315_EMITTER_PIN 4 //4 = D4 on arduino
#endif
```

Add the call to these functions into OpenMQTTGateway.ino 
Into the setup function
```
  #ifdef ZgatewayRF315
    setupRF315();
  #endif
```
Into the loop function
```
    #ifdef ZgatewayRF315
      if(RF315toMQTT())
      trc(F("RF315toMQTT OK"));
    #endif
```

And finally add the ZgatewayRF315 activation into user_config.h below ZgatewayRF
```
  #define ZgatewayRF315
  #include "config_RF315.h"
```

closed 2017-12-07 by 1technophile

jxjhheric, 2017-12-08:
@1technophile  thanks

NeoMatrixJR, 2018-01-10:
I followed the above...changed the e/r pins a bit, but now I'm running into an issue where RX on 315 comes across as 433toMQTT in topics and issuing commands to 315 via MQTT isn't working either trying to TX.  I'm thinking this may have something to do with mySwitch and mySwitch315 in the zgateways are = RCSwitch(); and are not instances so they're overlapping somehow?  has anyone else experienced this?

NeoMatrixJR, 2018-01-11:
@1technophile - Do the emitters have to remain on the same pin?  Won't that generate unnecessary transmissions on the opposite other frequency for every command?

1technophile, 2018-01-11:
@NeoMatrixJR nop It is better to have them on separate pin, same things for the receivers

NeoMatrixJR, 2018-01-11:
@1technophile - I put everything on separate pins (see previous comment...sorry I didn't tag you on it).  There seems to be an issue where all RX gets reported via 433toMQTT even if it's 315, and 315 tx doesn't seem to work.  I'm wondering if there's an issue that RCSwitch isn't running 2 instances and understanding that there's 2 sets of pins?