1technophile/OpenMQTTGateway #141: Support for RGB LED

Landrash, 2017-11-21:
Suggest support for RGB LED by the use of the http://fastled.io/ library.

Suggest integration with #40 for use as a status led.

posicat, 2018-01-15:
I've got a couple spools of these LEDs laying around.  One of my projects upcoming is to put a bunch of them under my kitchen cabinets.  I had intended to use an ATTiny and Raspberry Pi, but I don't see why this couldn't be built in as a feature to OpenMQTTGateway.  I've used the Adafruit libraries before, and one from Josh.com that's super-memory efficient.  On a project like this, it might be useful to have the small library available for a status LED, and the fastled library available for controlling whole strands.

posicat, 2018-01-15:
This is the other library I would investigate, if the fastled doesn't use too much memory, that sounds like a good candidate, this one however I run on ATTiny85's without memory problems, so it's nice and tiny.

https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/#more-1681

joen, 2018-01-23:
I would like to add fastled support. Is anybody working on this already?

posicat, 2018-01-23:
I had idly considered it, but I have plenty of work with the PCB project, and my own-home code that this all hooks into to get to adding LED support at this time.  I haven't seen anyone else mention taking this on, and my post revived a topic that had been idle 2 months, so it's probably safe.  Feel free to poke at me for any testing/help.  I've got plenty of WS2812's at home to try it with.

1technophile, 2018-01-23:
@joen I don't think so, in my point of view you can go

joen, 2018-01-23:
[first tests](https://github.com/joen/OpenMQTTGateway) with just one LED

1technophile, 2018-05-12:
thanks @joen could you rename to ZactuatorFASTLED (to follow the naming convention as we are acting on a device) and submit a pull request?

joen, 2018-05-29:
> could you rename to ZactuatorFASTLED
I am a little short on time, but will do it in the next days

QuagmireMan, 2018-06-19:
Very excited to see this feature!

insane2subro, 2018-08-01:
Very excited to see this work on LED Strips, FastLed is really amazing.

1technophile, 2018-09-07:
@joen do you think you could submit a pull request with your work?

joen, 2018-09-08:
submitted just now. #266 

This is all a litte dirty, but working.
You could theoretically use all RGB-LEDs, that are supported by fastled. Just change the configuration according to config_FASTLED.h

I am not quite happy with the effects, which could be done in a more intelligent manner.
The overall idea is to have a color value send to a topic that is used as the color the effect is "played" in and as the color that is set, when the effect is finished.

This was tested with just one WS2812 led (neopixel) on a ESP8266 (nodeMCU)

closed 2023-02-04 by 1technophile