1technophile/OpenMQTTGateway #142: Add IR raw signal forward

PetricaM, 2017-11-22:
Hi,

Could you add the option to forward the exact IR raw payload?

For IR devices using protocols with very large protocol length (ie. 112 bits) the payload is incompletely read (412 char although the total full raw message is 942 char).

Thanks

1technophile, 2017-12-02:
Hi, Could you test it from the master please (you need to put RawDirectForward true in config_IR.h)

PetricaM, 2017-12-02:
Hi Florian,

I've tried it and the IR payload is still 412 long. I think this is due to the buffer size on the NodeMCU for receiving as for both serial and MQTT connection the message has the same size, so it's not due to length of MQTT payload. Only receiving is affected as sending raw IR works with the full message length.

Thanks

1technophile, 2017-12-02:
Could you try to increase RAWBUF in irrecv.h ?

PetricaM, 2017-12-02:
Yes, that worked (I've set it to 300U).

Thanks

1technophile, 2017-12-02:
In this is case, is direct raw signal forward usefull or it can works also by mqtt signal forward (repeatIRwMQTT )

PetricaM, 2017-12-02:
The initial problem was the message length. 

It's perfectly fine to use MQTT payload to forward it to another device in HA.

1technophile, 2017-12-02:
Ok i will let the direct raw forwarding if someone need it

closed 2017-12-02 by 1technophile