1technophile/OpenMQTTGateway #144: IR Send Error

prasad7676, 2017-11-25:
Arduino: 1.8.3 (Mac OS X), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

/Users/prasad/Documents/Arduino/libraries/OpenMQTTGateway/ZgatewayIR.ino: In function 'void MQTTtoIR(char*, char*)':
ZgatewayIR:136: error: 'class IRsend' has no member named 'sendGC'
       irsend.sendGC(GC, j);
              ^
ZgatewayIR:188: error: 'class IRsend' has no member named 'sendCOOLIX'
     irsend.sendCOOLIX(data, valueBITS);
            ^
ZgatewayIR:201: error: 'class IRsend' has no member named 'sendWhynter'
     irsend.sendWhynter(data, valueBITS);
            ^
ZgatewayIR:208: error: 'class IRsend' has no member named 'sendLG'
     irsend.sendLG(data, valueBITS);
            ^
ZgatewayIR:237: error: 'class IRsend' has no member named 'sendSharpRaw'
     irsend.sendSharpRaw(data, valueBITS);
            ^
ZgatewayIR:259: error: 'class IRsend' has no member named 'sendRCMM'
     irsend.sendRCMM(data, valueBITS);
            ^
exit status 1
'class IRsend' has no member named 'sendGC'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

1technophile, 2017-11-25:
Could you give us the following details as asked in the submitting template:
* OpenMQTTGateway version used (V0.4, 0.3, master...)
* For IR and RF clarify if you tested with the basic examples given with these libraries
* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]

prasad7676, 2017-11-25:
thanks man for quick reply,

yes you was right has a libraries issue. 

FIXED....thanks a lot

closed 2017-11-25 by 1technophile