1technophile/OpenMQTTGateway #148: OpenMQTTGateway control KAKU with RFM69

jordanChalet, 2017-12-08:
Hi, what a great work ! 
It has easily allowed me to build a gateway between [cloudMqtt](https://www.cloudmqtt.com) (modifications in OpenMqttGateway.ino are needed for me, should I post them ?) and DIO switches (newKaku protocol).
My next step is to replace the analog receiver and the analog transmitter by the RF69M.

But I saw in the wiki that ZgatewayRF2 and ZgatewayRFM69 are not compatible. Does anybody have already configure the ZgatewayRFM69 to handle data from the NewRemoteTransmitter library ?
What changes are needed to achieve this ?

1technophile, 2017-12-09:
Hello Jordan,

What is the kind of modifications you have done?

I don't think someone already used newremoteswitch (zgatewayRF2) library with RFM69, you will need to do some research to do this.

Here is some inspiration for OOK but I don't know if we can transpose it to newremoteswitch:
https://github.com/kobuki/RFM69OOK

1technophile, 2018-01-07:
I'm going to close this issue, if you have news don't hesitate to reopen it or chat me on gitter

closed 2018-01-07 by 1technophile