1technophile/OpenMQTTGateway #155: Change/Support Building with PlatformIO

scotthernandez, 2018-01-10:
I, and I'm sure many others , don't use the arduino IDE (regularly, or at all) so it would be good to have a command-line only build process. It would also allow people to completely script the build process for those that don't want to, or need easier instructions, to get started.

Also, I expect switching to platformio will enable more option for CI and other integrations, but I'm no expert there.

I understand switching completely (and then moving from ino->cpp files, I would hope) might be a burden on the main developers so just supporting building would be great as a start.

broekema, 2018-01-14:
Alternatively a Makefile based build and install method could be an option. I've been playing with that for ESP8266, based on [this](https://github.com/plerup/makeEspArduino) work. Apart from some weirdness with code ordering (Arduino allows code to be used before it is declared, while C and C++ require forward declaration in such cases) it seems to be a viable alternative. If there is interest, I could prepare a pull request.

scotthernandez, 2018-01-20:
How open are we to switching away from using the arduino ide for building? I will work on the reorg needed for platformio and files needed if serious about using it from the main contributors.

1technophile, 2018-01-20:
Is it possible to switch and let also the support for arduino ide = keep ino files

scotthernandez, 2018-01-20:
Yes, I believe so, that would be my first suggested step. But I think the
files at least need to move into src dir and deps will be managed by pio
instead of linked into git. I'm not sure if that will allow arduino to
build anymore since I'm not familiar with it.

bukurat, 2018-01-23:
A number of the libraries are already set up for PlatformIO. Look for a PlatformIO.ini file in the root directory.
PlatformIO will compile .ino files so it is possible to build on either platform. PlatformIO doesn’t create function prototypes like the Arduino IDE so prototypes are needed. I have started setting up my fork of the program for PlatformIO.

scotthernandez, 2018-01-24:
Great, I'll hold off on working on this unless you want help.

bukurat, 2018-01-24:
The [tasmota](https://github.com/arendst/Sonoff-Tasmota/) project builds with either PlatformIO or the Arduino IDE, although he does recommend  separate Arduino IDE be set up for it.

My fork of this one is [here](https://github.com/bukurat/OpenMQTTGateway)  Scott, if you want to  help track down some compile errors.

thoscut, 2018-06-10:
Have you managed to sucessfully build with pio?

cnf, 2019-06-08:
I'm assuming no progress on this? I can't see any platformIO parts in the repo.

1technophile, 2019-07-16:
Currently in progress in platformio-dev branch

cnf, 2019-07-17:
ah, amazing! and just as platformIO went fully open, as well! <3 for the debugging! Thanks @1technophile !

1technophile, 2019-07-18:
integrated in development branch

closed 2019-07-18 by 1technophile