1technophile/OpenMQTTGateway #156: Publish features (via MQTT) for discovery

scotthernandez, 2018-01-11:
It would be nice if each device published information about its capabilities, and the topics for each, so that clients can automate discovery; for example, home-assistant could have a simple option in the config file that creates the appropriate sensor/ui based on discovery of the device (and one topic).

For example (after changing to json messages with simplified topics):
**topic**: `"home/" + Gateway_Name + "/features"`
**msg**: `{device: { id: XXX, ...}. temp: { topic: "home/DHTtoMQTT/dht/temp"}, humidity:
 {topic: "home/DHTtoMQTT/dht/hum"}, ... }`

scotthernandez, 2018-01-11:
TL;DR; -- Maybe, probably but not sure that homie should be implemented/supported, and json should be used instead.

Homie seems to be much more and also requires changing values in addition to defining metadata about the device/features and would seems to be much more work than just exposing metadata about the device. Also, it seems to create topics for each individual  (metadata) value, and doesn't seem to use structured-data/json payloads at all.

Is the homie convention the new plan for how things will be defined in mqtt?

I see #49, which seems to be logically incompatible with homie, imho -- seems like the system should be either json, or scalar values across many topics.

Personally, I would rather see a single topic with json docs rather than many topics with single (scalar) values for each of the attributes/metadata.

1technophile, 2018-01-11:
As you, I see the implementation of json as an opportunity to have auto discovery for controllers.
Nevertheless there is several ways to do auto discovery and Homie could be one.
But at final I don't know yet which way is the better one...
At least we miss a sort of convention/naming/syntax... around MQTT for interactions between nodes/gateway and controllers

scotthernandez, 2018-01-11:
I'd vote for json as it can be used independent of transport (http/mqtt/etc), it can be schema-validated and is easier to document/describe.

Yes, more than one can be supported ... but device memory is limited, as well as human resources for implementation, maintenance, configuration and support (testing including).

I'd suggest linked this to the json work, and not as a duplicate of homie.

1technophile, 2018-01-11:
I think as a first step we could add json and when there is consensus on a syntax/taxonomy on MQTT we go for it.
By schema validated you are thinking on something in particular?

1technophile, 2018-01-11:
@scotthernandez if I do the work on OpenMQTTGateway would you be interested to do the corresponding config for Home Assistant?

scotthernandez, 2018-01-11:
Yep

beed2112, 2018-04-05:
I believe that both these projects utilize json based mqtt discovery and integrate with Home Assistant
https://github.com/aderusha/MQTTCarPresence
https://github.com/aderusha/HASwitchPlate

For "local items" (PIR, Temp/Humidity, BT)  using this format 

`
#define Gateway_Name "LocationName"
`
To publish to the following topics:

```
home/<Gateway_Name>/motion
home/<Gateway_Name>/dht1
home/<Gateway_Name>/BTtoMQTT
```

Landrash, 2018-04-07:
Another project utilizing MQTT Discovery with Home Assistant is [ESPHomeLib](https://github.com/OttoWinter/esphomelib).

1technophile, 2018-05-05:
some info for ha https://www.home-assistant.io/docs/mqtt/discovery/

rjsachse, 2018-05-27:
Yes better with json. I have been working on a webserver with bootstrap using PROGMEM, so for sensor updates i am using json. I like the idea of discovery.

debsahu, 2018-06-12:
I implemented MQTT discovery in McLighting (https://github.com/toblum/McLighting/blob/6798a128b646a678a8835b0d7074e5b5cd5937d1/Arduino/McLighting/request_handlers.h#L965), let me know if you need help.

beed2112, 2018-06-13:
debsahu just upgraded to that version of mclighting. Very nice work one the mqtt ha piece

1technophile, 2018-06-18:
this feature has been identified as the top priority by the users:
https://community.openmqttgateway.com/t/poll-results-about-the-next-features-for-openmqttgateway/30

1technophile, 2018-06-28:
@debsahu do you need some inputs from me so as to progress on this ?

debsahu, 2018-06-28:
@1technophile sorry I was caught up with personal stuff. I will find some time this weekend and get this going. Is the email on GitHub the best way to get in touch with you?

1technophile, 2018-06-28:
Maybe best way is to open a topic on community.openmqttgateway.com and pm me on it if necessary

1technophile, 2018-09-18:
@debsahu hello, are you still planning to participate on this request?

animavitis, 2018-11-13:
im implementing Home Assistant mqtt discovery plugin right now.
I think today will be able to push first version with OMG status binary sensor & sensors for BME280

1technophile, 2018-11-13:
great!!!

Landrash, 2018-12-13:
> all sensors added here
> based on @animavitis <https://github.com/animavitis> work.
> Nevertheless I would like to have your ideas for gateway auto discovery.
> Do we create one switch for each signal received? Or do we create one
> sensor per gateway that receive signals and you do automations from it?
> As I'm not an HA expert I need your insights.

One sensor for gateway state and one for each switch makes most sense.

1technophile, 2018-12-16:
#323 
First implementation merged in master
The sensors has beed added
The gateway as sensors has been added
For the moment only the RF gateway create switch automaticaly, I wait for your feedback to see if it answer your needs before duplicating it. 
At the moment I don't publish or take into account advanced parameters like protocol/delay/bits for switches. I will need help for that as I'm not sure if home assistant can create a switch with several json attributes to be taken into account (it seems that json attributes parameter is only for sensors)

closed 2019-01-31 by 1technophile