1technophile/OpenMQTTGateway #159: ZgatewayBT:78: error: 'class BLEAdvertisedDevice' has no member named 'haveServiceData'

rolfberkenbosch, 2018-01-12:
I get an errror with by ESP32 with BLE support when i verify the code in arduino. How can I fix this ?

Documents\Arduino\OpenMQTTGateway\ZgatewayBT.ino: In member function 'virtual void MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice)':

ZgatewayBT:78: error: 'class BLEAdvertisedDevice' has no member named 'haveServiceData'

             if (advertisedDevice.haveServiceData()){

                                  ^

ZgatewayBT:80: error: 'class BLEAdvertisedDevice' has no member named 'getServiceData'

                 std::string serviceData = advertisedDevice.getServiceData();

                                                            ^

ZgatewayBT:94: error: 'class BLEAdvertisedDevice' has no member named 'getServiceDataUUID'

                 BLEUUID serviceDataUUID = advertisedDevice.getServiceDataUUID();

                                                            ^

1technophile, 2018-01-22:
I close the issue, reopen it if you need

closed 2018-01-22 by 1technophile

delta010, 2019-09-30:
I am using esp32 I have the same error,  what should I do

1technophile, 2019-09-30:
This is surely linked to a bad library version, you should use the versions linked here:
https://github.com/1technophile/OpenMQTTGateway/blob/e4de45f2194fa1fe30b2f4e51d6acdc1925802be/platformio.ini#L54