1technophile/OpenMQTTGateway #160: Gateways should have unique-name paths as well for sending commands

posicat, 2018-01-18:
So as to have a more efficient treatment of your request please provide the following information:
* OpenMQTTGateway version master

I have 2 gateways, creatively named OMG_001 and OMG_002, but when I send an IR broadcast command I send it to home/commands/IR_GC which both modules receive.  That's fine, but it would be great to have a way to send directly to a specific device like /home/OMG_001/commands/IR_GC to send the message to just one of the devices.  This would help in the case where a household has 2 of the same brand TV or receiver.

posicat, 2018-01-20:
Just a quick note, it occurred to me that you don't need the global-address since MQTT handles that nicely.  if all the devices have their names in the path /home/OMG_001/commands/IR_GC  you can address all of them as : /home/+/commands/IR_GC.  That might save a tiny bit of ram not having to register both formats.

Once I get a bit more familiar with the code layout I want to jump in and help code some things as well.  Right now just getting my one module working with my home system.

broekema, 2018-01-21:
This really needs to be propagated through the entire code. I've done this in my private repository by replacing the home/ occurrences in the code with something like this:

`#define TEMPBMEC MQTT_ROOT Gateway_Name  "/bme/tempc"
#define TEMPBMEF MQTT_ROOT Gateway_Name  "/bme/tempf"
#define HUMBME   MQTT_ROOT Gateway_Name  "/bme/hum"
#define PRESSBME MQTT_ROOT Gateway_Name  "/bme/pa"
#define ALTIBMEM MQTT_ROOT Gateway_Name  "/bme/altim"
#define ALTIBMEFT MQTT_ROOT Gateway_Name "/bme/altift"
`

closed 2018-04-04 by 1technophile