1technophile/OpenMQTTGateway #162: RCswitch receive 433mhz not working on pin 13

betonishard, 2018-01-19:
Hello,

First of all thank you for your effort. This is great.

I am building a gateway which I will use for a single mqtt hub. However I have a problem with RF reception. I am using a MH-ET D1 esp32. Here is the link for the board.
http://mh.nodebb.com/assets/uploads/files/1497417590890-1111.jpg?v=hfe2ofo3e00

I tried setting it up on TCK pin (GPIO13) and commented everything in the code. So only RF reception and sending. ALso tried GPIO17 and 27. I disconnected the transmitter just to test, however it does not come up in my serial monitor (ARDUINO IDE 1.8.5) .. I can see that Zgateway RF setup is done and the receiver pin is set to 13. However whatever I try I don't receive codes.  I rewired everything to a esp8266 (nodemcu) and it was working on that one. So I know the receiver functions.

Can you point me in the right direction please?

Keep up the good work..
Regards!

bandric1, 2018-01-20:
Im Experiencing same issue ESP32 DevkitV1 on D13 doesnt work the RF 433 reader.

1technophile, 2018-01-20:
Yes my doit esp32 board supply 5v, maybe not the case for all boards

bandric1, 2018-01-20:
It is strange made it fire twice or three times than it stoped.

betonishard, 2018-01-21:
Thank you for your reply.

I didn't mention it in my first post, but I have a level converter between the data pin of the receiver and the ESP. So pin gpio 13 is connected to for example LV1, and the receiver on HV1. I am supplying the receiver with 5v from a 5v powersupply. 

Any suggestions?

Thank you for your reply.

Best regards,

betonishard, 2018-01-21:
EDIT 2: Maybe someone has another pin in use and could share what pin is the second best solution after pin 13..?
Thanks in advance.
Best regards.

bandric1, 2018-01-21:
i have tried all the other remaining pins addressing to them still didnt work. IR module is working normally to me looks like RF code is somehow not compatible with my version of esp32. I dont know how it fires for three times and displayed the code in mqtt but it did and i cant replicate the event no matter what i do.

betonishard, 2018-01-22:
@bandric1 sorry to hear that.. Maybe @1technophile has an idea to try? I am out of options...

1technophile, 2018-01-22:
Did you took the last version of ESP32 arduino?
You can also try to power a LED+ resistor (or only a resistor) on the pin used to power up your receiver and measure the voltage with a multimeter.

I tried these pins on my DOIT ESP32 board:
13 OK
27 OK
26 KO
25 KO
33 KO

If not working I suggest to try with another ESP32 board.

betonishard, 2018-01-22:
Hi, thank you for reply.

I don't understand why i should use a led or resistor. If I measure just with multimeter the power on my receiver and see 5 volt wouldn't it be enough? Could you clarify please?

Thanks

1technophile, 2018-01-23:
It could be that the voltage drop down when a charge is plugged to the power supply, that s why i ask to add a resistor.
Here is the link for arduino esp32 repo : 
https://github.com/espressif/arduino-esp32

1technophile, 2018-02-08:
@betonishard did you finaly get it working?

betonishard, 2018-02-09:
@1technophile I finally couldn't solve it so I bought the Doit version just for testing. I was able to test this, however pin 13 still wasn't working. GPIO 27 worked directly. When I did that my BLE was completely gone (really strange). The same issue people mentioned in one of the latest posts. I set the priority to 0 and it started working. Haven't been able to do a duration test yet. But I think I will this weekend.

Regards

1technophile, 2018-04-04:
wiki updated to include pin 27

closed 2018-04-04 by 1technophile

Luc3as, 2018-10-17:
Guys I am trying to solve my issue, I've got ESP32 working with RF SRX882 without problem through pin 27, however after some time It stopped working and now I am unable to find solution, I tried another RF receiver, older firmware, still nothing. I am powering RF from VIN on doit ESP32, and my question is,  could I fried GPIO at ESP because RF is running at 5v from charger and esp has 3v3 logic ?  I  don't have spare ESP32 to try but now the only thing I did not change is ESP itself. what's your scheme for connecting this are you using some resistor on data pin ? thank you