1technophile/OpenMQTTGateway #164: String handling improvment

1technophile, 2018-01-22:
Following this discussion improve string handling on the projects
https://github.com/1technophile/OpenMQTTGateway/pull/154

broekema, 2018-01-23:
Thinking about this, we could probably generalize the publishing of messages into a global function or method used by all other code. In this function we can then differentiate between Uno and ESP targets, but only once and at a well defined central location.

1technophile, 2018-01-23:
Great idea !

posicat, 2018-01-23:
This is where a JSON library could come in handy, might want to evaluate that and this topic together as one change.

1technophile, 2018-01-23:
Good point @posicat, we might first try to include json publishing before or in the same time of  that

broekema, 2018-01-28:
I've create pull request #168 for discussion. This is a class based sensor refactor, that centralizes publishing of messages.

1technophile, 2023-02-04:
Implemented arduinolog for traces, and centralized pub functions

closed 2023-02-04 by 1technophile