1technophile/OpenMQTTGateway #165: [FEATURE REQUEST] Support decoder and parser for IKEA Sparsnas electricity meter?

Hedda, 2018-01-23:
Would it be possible to add this decoder for IKEA Sparsnäs (Ikea Sparnas electricity meter)?

There is already an existing project that does this with ESP8266 & RFM69 but nothing more:

https://github.com/bphermansson/EspSparsnasGateway

Some more information and links are posted here which I copy below:

http://www.letscontrolit.com/forum/viewtopic.php?f=10&t=3726

+1 request for IKEA Sparsnäs (Ikea Sparsnas) electricity meter with wireless display

http://www.ikea.com/se/sv/catalog/products/90292645/

At least in Sweden these are often on sale every so often with a large rebate via IKEA FAMILY and as such they can also be found relativly cheap from rebate scalpers on eBay (and its Tradera action-site in Sweden) following closely aften a sale.

http://www.ikea.com/se/sv/manuals/sparsnas-energidisplay__AA-1201480-1.pdf

"[i]SPARSNÄS energy display assists you to monitor the total usage of electricity in your home. The energy display consists of a transmitter and a display. The transmitter, which you attach to your electricity meter, continuously registers the usage of electricity and transfers the data wireless to the display. Place the display where it becomes a natural part of daily life in your home"

Again, there is an open source RTL-SDR decoder and parser for it:

https://github.com/strigeus/sparsnas_decoder

"SPARSNÄS uses a CC1101 chip configured for GFSK modulation at 868MHz. The two FSK symbol frequencies once downconverted to baseband are roughly 67kHz and 105kHz. (EDIT: It seems like with a different RTL-SDR the baseband frequencies are 12.5khz and 50khz. Maybe my first RTL-SDR was inaccurate). The packet length is 18 bytes, including the length byte, and a 16-bit CRC is appended to the end. The CC1101 sync word is 0xD201. The packet payload byte 5 to 17 are encrypted using a key derived from the sender device's ID. The encryption key is a repeating XOR key"

tubalainen also has a fork of the above RTL-SDR project which convert and output as JSON to be used to pipe data to MQTT broker:

https://github.com/tubalainen/sparsnas_decoder

"The packet data is encrypted using your sender's ID. The sender ID is the last 6 digits of the serial number located under the battery."

"A BIG thanks to the https://www.facebook.com/groups/SHgruppen on Facebook"

Hedda, 2018-01-25:
To reverse this suggestion;  here is code projects to instead send a signal to Sparnas wireless display:

https://github.com/MagnusThome/sparsnas-esp8266-01

https://github.com/MagnusThome/sparnas-raspberry

posicat, 2018-01-25:
I did some research over lunch on how the IKEA Sparnas works. It reads an LED on the power-meter that blinks every time a specific number of watt hours are used.

It would be far cheaper to build in functionality to OpenMQTTGateway that could read the LED pulses than to worry about interfacing to the Sparnas. I hope to try experimenting with my own meter this weekend.  However for people who already have a Sparnas or are thinking of getting one, it would be nice to have code to read it.

Hedda, 2018-01-30:
@posicat cheaper yes but not userfriendly to those who already own an Ikea Sparnäs and have it installed.

Hedda, 2018-01-30:
FYI; here are more detailed information information on how exactly Sparnäs work with power-meters :

https://github.com/kodarn/Sparsnas

@kodarn include a "LedBlinkerHelperToo" to emulate power-meter to help develop without Sparnäs.

Other people have also created similar LED blinkers for for it to do send a value to Sparnäs display, ex: 

https://github.com/MagnusThome/sparsnas-esp8266-01
https://github.com/MagnusThome/sparnas-raspberry

posicat, 2018-01-30:
@Hedda I pointed out that we could build a similar interface easily just for those interested in monitoring their meters, that the Ikea Sparsnas wasn't the only option.  Especially for people like me in the US who's Ikea sadly doesn't sell them.

If you have one and are enjoying it, absolutely keep using it and add support.  At the same time the code to support the Sparsnas could include code to use a sensor directly.  Then both Sparsnas and Makers could get their results in the exact same format.

1technophile, 2023-02-04:
@NorthernMan54 could this sensor using FSK (compatible rtl_433) be read with OMG your library and a CC1101?