1technophile/OpenMQTTGateway #166: Port-setup section in OpenMQTTGateway.ino

posicat, 2018-01-28:
Having to go to each individual .h file to configure ports is a bit inconvenient, I was wondering perhaps moving them to User_config.h might be possible.  That way every configuration for a specific module I might have would reside in one file, including the #defines that turn that module on.

A couple benefits of this:
1) I could make multiple config files, and editing one line of OpenMQTTGateway.ino would include the settings, *.h files and whatnot I need for that specific board I'm re-programming.
2) The other .h files would remain unchanged when changing configurations, making troubleshooting when something doesn't work a great deal simpler.

bphermansson, 2018-02-09:
I'm thinking the same. Also the Mqtt-topics should be in one place, now it's a pain to look around for different settings.

1technophile, 2018-02-18:
There is pro and cons,
Sol A one file for pinout: pro yes it can be easier to have all the pins in one place in particular if you search the "ideal" pinout
Sol B separate files for pintout: cons in a developper point of view to have a file for each sensor/gateway avoid conflict during merge

but maybe pros of sol A is more important than cons of sol B, it's open, if others have ideas to share about this?

1technophile, 2018-04-05:
@posicat Sorry maybe I didn't understand well at my first read, are we talking about only centralizing the pinout or all the setup sections?

posicat, 2018-04-05:
It really could go either way, the main could just call into the sensor modules with the appropriate parameters.  I don't have the code in front of me, but I could see something like

#define RF_TX_PIN 4
#define RF_RX_PIN 5
//#define USE_RF_MODULE

Where the user could quickly change the pin number and enable the module at the same time.

1technophile, 2018-07-28:
I'm thinking about centralizing the pinout as proposed but on a dedicated file like PINOUT_config.h, We could also separate the credentials into a dedicated file. Doing this will simplify the maintenance and aboid the stress of publishing its credentials into the web.

1technophile, 2019-04-15:
I'm closing this issue as I have no more requests on it

closed 2019-04-15 by 1technophile