1technophile/OpenMQTTGateway #169: Sonoff RF bridge reconnect frequently

mizrachiran, 2018-01-31:
Hi,

I'm using the latest version on a Sonoff RF Bridge. 
It works great but it seems like it reconnect once in ~20 minutes. (I see offline and online messages sent over LWT topic).

Why is that happening? 

[image]

mizrachiran, 2018-02-01:
just noticed on my MQTT server (Mosquitto) the following logs:

"Socket error on client OpenMQTTGateway"

1technophile, 2018-03-09:
Are you sure you don't have another OpenMQTTGateway with the same name on the same network?
If not please try with the last master version

mizrachiran, 2018-03-09:
I have only 1 OpenMQTTGateway installed.

mizrachiran, 2018-03-09:
I am using the latest version

1technophile, 2018-03-10:
Did you updated your broker version ?  Which one are you using

mizrachiran, 2018-03-10:
i'm using v1.0.

1technophile, 2018-03-10:
If you are using mosquitto you should update it

mizrachiran, 2018-03-10:
OK. I will give it a try.
Closing this thread for now.

Thanks!

closed 2018-03-10 by mizrachiran