1technophile/OpenMQTTGateway #172: Using Widora Air (ESP32) - Can't compile with BLE turned on in config.

NeoMatrixJR, 2018-02-04:
So as to have a more efficient treatment of your request please provide the following information:
* OpenMQTTGateway version used MASTER
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library) - CURRENT BLUETOOTH LIBRARY - did mods listed on wiki

is this normal to have so many multiple definitions?  I can't get it to compile.

compile log:
`Arduino: 1.8.5 (Windows 10), Board: "Widora AIR, 80MHz, 115200"

Archiving built core (caching) in: C:\Users\neoma\AppData\Local\Temp\arduino_cache_707431\core\core_espressif_esp32_widora-air_FlashFreq_80,UploadSpeed_115200_898f9e38ebae302494d0e119be95335d.a
libraries\BLE\BLE2902.cpp.o: In function `BLE2902::BLE2902()':

C:\Program Files (x86)\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLE2902.cpp:17: multiple definition of `BLE2902::BLE2902()'

libraries\ESP32_BLE_Arduino\BLE2902.cpp.o:C:\Users\neoma\OneDrive\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLE2902.cpp:17: first defined here

libraries\BLE\BLE2902.cpp.o: In function `BLE2902::BLE2902()':

BLE2902.cpp:(.text._ZN7BLE2902C2Ev+0x0): multiple definition of `BLE2902::BLE2902()'

libraries\ESP32_BLE_Arduino\BLE2902.cpp.o:BLE2902.cpp:(.text._ZN7BLE2902C2Ev+0x0): first defined here

libraries\BLE\BLE2902.cpp.o: In function `BLE2902::getNotifications()':

BLE2902.cpp:(.text._ZN7BLE290216getNotificationsEv+0x0): multiple definition of `BLE2902::getNotifications()'

libraries\ESP32_BLE_Arduino\BLE2902.cpp.o:BLE2902.cpp:(.text._ZN7BLE290216getNotificationsEv+0x0): first defined here

libraries\BLE\BLE2902.cpp.o: In function `BLE2902::getIndications()':

BLE2902.cpp:(.text._ZN7BLE290214getIndicationsEv+0x0): multiple definition of `BLE2902::getIndications()'

libraries\ESP32_BLE_Arduino\BLE2902.cpp.o:BLE2902.cpp:(.text._ZN7BLE290214getIndicationsEv+0x0): first defined here

libraries\BLE\BLE2902.cpp.o: In function `BLE2902::setIndications(bool)':

BLE2902.cpp:(.text._ZN7BLE290214setIndicationsEb+0x0): multiple definition of `BLE2902::setIndications(bool)'

libraries\ESP32_BLE_Arduino\BLE2902.cpp.o:BLE2902.cpp:(.text._ZN7BLE290214setIndicationsEb+0x0): first defined here

libraries\BLE\BLE2902.cpp.o: In function `BLE2902::setNotifications(bool)':

BLE2902.cpp:(.text._ZN7BLE290216setNotificationsEb+0x0): multiple definition of `BLE2902::setNotifications(bool)'

libraries\ESP32_BLE_Arduino\BLE2902.cpp.o:BLE2902.cpp:(.text._ZN7BLE290216setNotificationsEb+0x0): first defined here

libraries\BLE\BLE2904.cpp.o: In function `BLE2904::BLE2904()':

C:\Program Files (x86)\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLE2904.cpp:18: multiple definition of `BLE2904::BLE2904()'

libraries\ESP32_BLE_Arduino\BLE2904.cpp.o:C:\Users\neoma\OneDrive\Documents\Arduino\libraries\ESP32_BLE_Arduino\src/BLE2904.cpp:18: first defined here

libraries\BLE\BLE2904.cpp.o: In function `BLE2904::BLE2904()':

BLE2904.cpp:(.text._ZN7BLE2904C2Ev+0x0): multiple definition of `BLE2904::BLE2904()'

libraries\ESP32_BLE_Arduino\BLE2904.cpp.o:BLE2904.cpp:(.text._ZN7BLE2904C2Ev+0x0): first defined here

libraries\BLE\BLE2904.cpp.o: In function `BLE2904::setDescription(unsigned short)':

BLE2904.cpp:(.text._ZN7BLE290414setDescriptionEt+0x0): multiple definition of `BLE2904::setDescription(unsigned short)'

libraries\ESP32_BLE_Arduino\BLE2904.cpp.o:BLE2904.cpp:(.text._ZN7BLE290414setDescriptionEt+0x0): first defined here

libraries\BLE\BLE2904.cpp.o: In function `BLE2904::setExponent(signed char)':

BLE2904.cpp:(.text._ZN7BLE290411setExponentEa+0x0): multiple definition of `BLE2904::setExponent(signed char)'

libraries\ESP32_BLE_Arduino\BLE2904.cpp.o:BLE2904.cpp:(.text._ZN7BLE290411setExponentEa+0x0): first defined here

[... 2625 lines cut ...]
GeneralUtils.cpp:(.text._ZN12GeneralUtils7toLowerERNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): multiple definition of `GeneralUtils::toLower(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'

libraries\ESP32_BLE_Arduino\GeneralUtils.cpp.o:GeneralUtils.cpp:(.text._ZN12GeneralUtils7toLowerERNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): first defined here

libraries\BLE\GeneralUtils.cpp.o: In function `GeneralUtils::trim(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':

GeneralUtils.cpp:(.text._ZN12GeneralUtils4trimERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): multiple definition of `GeneralUtils::trim(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'

libraries\ESP32_BLE_Arduino\GeneralUtils.cpp.o:GeneralUtils.cpp:(.text._ZN12GeneralUtils4trimERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): first defined here

libraries\BLE\GeneralUtils.cpp.o: In function `GeneralUtils::split(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, char)':

GeneralUtils.cpp:(.text._ZN12GeneralUtils5splitENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEc+0x0): multiple definition of `GeneralUtils::split(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, char)'

libraries\ESP32_BLE_Arduino\GeneralUtils.cpp.o:GeneralUtils.cpp:(.text._ZN12GeneralUtils5splitENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEc+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Widora AIR.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

1technophile, 2018-02-04:
I relaunched a build to see if esp32 env update makes the build broken, you can follow the result here:
https://travis-ci.org/1technophile/OpenMQTTGateway/jobs/336738178

if it is ok, your issue is most likely due to a bad component version

1technophile, 2018-02-04:
the components used for this build can be found here:
https://github.com/1technophile/OpenMQTTGateway/blob/master/.travis.yml

1technophile, 2018-02-04:
Indeed the link given in the library folder is not good the last version of ESP32_BLE_Arduino doesn't compile, I confirm that you can take this one and it compile:
https://github.com/nkolban/ESP32_BLE_Arduino/archive/f8fe9d7cdfb20caa54b70849826d1ac6e375ff78.zip

NeoMatrixJR, 2018-02-05:
@1technophile - I grabbed that last zip you posted and dropped it in place.  Do I still need to modify the files with the ones on the wiki?  Just straight dropping in the zip produced this:
[Recompile pastebin with new zip](https://pastebin.com/v31JRQ1Z)

1technophile, 2018-02-05:
Yes you have to overwrite the 2 files

NeoMatrixJR, 2018-02-06:
works!

closed 2018-02-06 by NeoMatrixJR