1technophile/OpenMQTTGateway #173: BLE on ESP32 is not working anymore after a few hours

rolfberkenbosch, 2018-02-04:
Hi All,

Does anybody know's why the BLE on the ESP32 is not working anymore after a few hours. I need to turn off the power and power it again.  Below the messages when it is not working anymore:

Data stored
E (4452082) BT: btc_search_callback  BLE observe complete. Num Resp 0

BT Task running on core 0
E (4482287) BT: btc_search_callback  BLE observe complete. Num Resp 0

BT Task running on core 0
E (4512492) BT: btc_search_callback  BLE observe complete. Num Resp 0

BT Task running on core 0

When it is working i see the following:

BT Task running on core 0
Get ManufacturerData 

Get RSSI 
home/BTtoMQTT/3165909E9745 -47
Get ManufacturerData 
⸮
Get RSSI 
home/BTtoMQTT/7E8538567669 -81
Get service data 
0000000000000000000000000000000000000000
Get service data UUID
0000fe9f-0000-1000-8000-00805f9b34fb
Get ManufacturerData 
⸮
Get RSSI 
home/BTtoMQTT/57FB4DB4D297 -68
Get service data 
0000000000000000000000000000000000000000
Get service data UUID
0000fe9f-0000-1000-8000-00805f9b34fb
Get Name 
Flower care
Get RSSI 
home/BTtoMQTT/C47C8D661011 -87
Get service data 
71209800821110668d7cc40d0710034e0000
Get service data UUID
0000fe95-0000-1000-8000-00805f9b34fb
Processing mi flora data

Does anybody has a fix for this or know how to solve this ?

1technophile, 2018-02-04:
Hello, 

I'm investigating on this issue, the same issue was reported on home assistant forum

rolfberkenbosch, 2018-02-04:
Ok, let me know if I can do anything for you to help.

1technophile, 2018-02-04:
thanks for proposing, could you change l117 of ZgatewayBT and replace 1 by 0
Current
```
                          1,          /* Priority of the task */
```
New
```
                          0,          /* Priority of the task */
```

and try again

In the meanwhile I'm doing another test

rolfberkenbosch, 2018-02-05:
Hi, I have tried that but the same problem accour after a few hours._

alpat59, 2018-02-07:
I’m having the same problem, also after the change l117
Thank you in advance for a fix

betonishard, 2018-02-12:
Hi all, using a doit esp32 I have the same problem, after a few hours the board becomes totally unresponsive. When you are using the BLE options other components fail as well (RF433).
Thanks in advance

Boxter83, 2018-02-13:
I`m also have the same problem with ESP32 that i lose connection efter som time. I tried to find cause, but right now i have only find out that it is the wifi connection that fails after some time. Tried to ping the IP adress but no answer, but i can see in the seriall monitor that everyting else works.

1technophile, 2018-02-14:
Thanks for the infos, i will take a look on it in a few days

1technophile, 2018-02-16:
It seems that there is a memory leak somewhere, i have the available ram decreasing continuously.
I have to isolate now where it comes from.

1technophile, 2018-02-17:
@all could you test last version from master please, I did a modification after @nkolban advices

betonishard, 2018-02-17:
@1technophile replaced the ZgatewayBT.ino file in my current setup and test is running.

I will post my findings

Boxter83, 2018-02-17:
I tryed the new ZgatewayBT.ino but then it dosen't finde my Miflora anymore. Anymore else how made it work better than me?

alpat59, 2018-02-17:
I tried too and it doesn't find BLE devices. Thank you for your support

alpat59, 2018-02-17:
sorry .. the behaviour changed just now after 3/4 reboot (why?). Now I can see BLE devices. I'll check if it will work stable.

betonishard, 2018-02-17:
Hello @all. Actually it works good now in my setup. Maybe if I would share my setup it could be helpfull to others. I have a DOIT ESP32, with an RF receiver connect to gpio 27 and a rf sender connected to gpio 12. I have 2 bluetoothe beacons (nut) and they get detected as well as my miFlora is working also properly sinds this morning (for 7 hours straight now).

1technophile, 2018-02-17:
@alpat59 you mean with the new code it didn't find your beacons or it is the first time you tried with openmqttgateway. If you are in the second case you should better go to the chat room and gives infis about your setup

1technophile, 2018-02-17:
@alpat59 haven t seen your last post. Forget my comment.

rolfberkenbosch, 2018-02-18:
@1technophile I'm trying it also right now. There is indeed a problem with the wifi connection. I have made a ugly fix fix for it. I will push the fix to git tomorrow.

1technophile, 2018-02-18:
@betonishard are you also having a problem with the wifi connection ?

betonishard, 2018-02-18:
@1technophile this morning it stopped working.. it worked close to 24 hours without problems. I don't know if it was wifi because I didn't hook it up to a serial monitor yesterday.

alpat59, 2018-02-18:
I’m having the same WiFi connection problem. Disconnection after some hours.

1technophile, 2018-02-18:
Maybe we should monitor this issue:
https://github.com/espressif/arduino-esp32/issues/653

1technophile, 2018-02-18:
I 've just added wifi auto reconnection

1technophile, 2018-02-20:
Do you have some feedbacks with auto reconnection added?

betonishard, 2018-02-21:
@1technophile it failed again. I am not sure about the reason, it wasn't connected to serial monitor when it happened. It occured after several hours.

betonishard, 2018-02-22:
@1technophile, the wifi part is working.. So the commands send out to the rf sender are working. Wifi is not down, however no bluetooth activity... (maybe priority 0, as suggested in earlier post?)

1technophile, 2018-02-22:
@betonishard do you see something on the serial monitor related to bluetooth?

I'm reproducing the same but I'm not connected to serial.

betonishard, 2018-02-22:
@1technophile. Mine is not connected to serial monitor either. I noticed it on mqttdash with my phone that ble is dead.

I have 2 but beacons and a miflora. No messages at all.. hope you can check it

1technophile, 2018-02-22:
I setup one esp32 with minimal BLE scan example with a serial monitor connection to monitor it. Feedback in a few days

rolfberkenbosch, 2018-02-23:
I will test it also, i will let you know the outcome.

1technophile, 2018-03-01:
Now working for more than 48 hours with the last version of master code

1technophile, 2018-03-02:
4 days without issues.

Could somebody confirm it is working continuously with the last master version?

betonishard, 2018-03-03:
@1technophile it seems to be working when it is started, however it still unpredictable. Sometimes when I reset the board it won't start, sometimes it will. When I connect a rf433 receiver on pin27 or pin 13. I plug in the the power I get "Guru Meditation Error: Core 1 panic'ed (Cache disabled but cached memory region accessed)". Bluetooth stops working and when I restart the board bluetooth won't work. When I wait a while (usually of couple of minutes) do some resets it starts working. When I  connect the 433 recevier when it's powered on, I won't get the guru error. So it's weird and kind of unpredictable. At least I couldn't find a pattern other than I just described.

1technophile, 2018-03-03:
Are you using my last commited version?

betonishard, 2018-03-03:
@1technophile Yes I am....

1technophile, 2018-03-03:
Could you send me your code and config files please and give me the board you are using?

What is your ESP32 arduino update date?

betonishard, 2018-03-03:
@1technophile what part of the code do you need? Everything, all files? How can I find my esp32 arduino update date? Sorry for the questions..

1technophile, 2018-03-03:
@betonishard you can send me all.
regarding ESP32 ide if you can remenber hen you donwloaded it

betonishard, 2018-03-03:
@1technophile, you can download it from here, https://github.com/betonishard/testing/blob/master/setup.zip

just did a pull on the esp32 
   52d4705..d29cfdb  master              -> origin/master
 * [new branch]      enhance-test-detail -> origin/enhance-test-detail
Updating 52d4705..d29cfdb
Fast-forward
 boards.txt                     | 52 +++++++++++++++++++++++++++++++++++++++
 docs/platformio.md             | 17 ++++++-------
 tools/platformio-build.py      |  5 ++--
 variants/pico32/pins_arduino.h | 56 ++++++++++++++++++++++++++++++++++++++++++
 4 files changed, 119 insertions(+), 11 deletions(-)

thanks for checking!

1technophile, 2018-03-04:
Could you try please to:
-download the last arduino ESP32 boards version
-comment in user_config.h ZgatewayRF, ZgatewayRF2 and config_rf.h

and give me feedback

1technophile, 2018-03-04:
when you are copying the nkolban folder are you taking this revision?
https://github.com/nkolban/ESP32_BLE_Arduino/tree/3f5005c383540d07dd16ac044c78d7be28d4aa8e

1technophile, 2018-03-04:
@rolfberkenbosch could you test also with the lateste revision of OpenMQTTGateway master please?

1technophile, 2018-03-04:
I just published all the libraries that I'm using, please take ESP32 BLE from this folder:
https://github.com/1technophile/OpenMQTTGateway/tree/master/lib

betonishard, 2018-03-04:
I used all relevant libraries. It seems to be more stable now.

I uncommented the RF, RF2 and config BLE was not working. With comment it it was working . Even when I did a powercycle / reset etc.. When I uncommented, it wasn't working anymore.

1technophile, 2018-03-04:
could you just uncomment RF and not RF2?

betonishard, 2018-03-04:
Did that already.. ;). Just to check. It did work. Now I have uncommented RF again and commented out
 remotereceiver in RF2 because I am not using it. I have cheap chinese receviers only and kaku plugs. It is now working for more than an hour stable including BLE. No guru meditation error either. FIngers crossed

betonishard, 2018-03-09:
@1technophile it worked with th setup above for 4 days consequtively.. i was getting the impression that it was stable. However on the fifth day.. ble stopped working. Rf still hoing strong but no ble.. i didn't hook it up to a serial monitor so i donT know why it stopped working..

I disconnected power, connected again. Ble was not coming up. I did a button reboot.. no ble. Any suggestions or any one else experiencing this as well?

1technophile, 2018-03-10:
Hello,

I'm going to setup again a basic sample of BLE scan linked to a serial monitor and see if it last more thant 4 days.
Alternatively could you try without RF gateway to see if it last longer?

alpat59, 2018-03-10:
@1technophile, I’m experimenting exactly what you described. I’m using ble only. Thank you for your great support.

1technophile, 2018-03-11:
@rolfberkenbosch could you try to increase the #define Scan_duration to 20 in config_BT.h with the current master code and see if it last longer than 4 days.

@alpat59 could you test please the modifications below with BLE only and gives us feedback how much it last,

The goal is to remove the BLE from being a task assigned to core 0 and to run it as a normal process with a millis condition when we don't use RF.

try to:
**In OpenMQTTGateway.ino replace in line 589:**
```
    #ifdef ZgatewayBT
      #ifndef ESP32
        if(BTtoMQTT())
        trc(F("BTtoMQTT OK"));
      #endif
    #endif
```
by 

```
    #ifdef ZgatewayBT
        if(BTtoMQTT())
        trc(F("BTtoMQTT OK"));
    #endif
```

**and replace all the content of ZgatewayBT by this one:**
```
/*  
  OpenMQTTGateway  - ESP8266 or Arduino program for home automation 

   Act as a wifi or ethernet gateway between your 433mhz/infrared IR signal/BLE  and a MQTT broker 
   Send and receiving command by MQTT
 
  This gateway enables to:
 - publish MQTT data to a different topic related to BLE devices rssi signal

    Copyright: (c)Florian ROBERT
  
    This file is part of OpenMQTTGateway.
    
    OpenMQTTGateway is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    OpenMQTTGateway is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.

Thanks to wolass https://github.com/wolass for suggesting me HM 10 and dinosd https://github.com/dinosd/BLE_PROXIMITY for inspiring me how to implement the gateway
*/
#ifdef ZgatewayBT

  #ifdef ESP32
    /*
       Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
       Ported to Arduino ESP32 by Evandro Copercini
    */
    // core task implementation thanks to https://techtutorialsx.com/2017/05/09/esp32-running-code-on-a-specific-core/
    
    #include <BLEDevice.h>
    #include <BLEUtils.h>
    #include <BLEScan.h>
    #include <BLEAdvertisedDevice.h>

    //Time used to wait for an interval before resending BLE infos
[... 267 lines cut ...]
    break;
    case '4' :
          mactopic = mactopic + "/" + "tem";
          if (value > 65000) value = value - 65535;
          dtostrf(value/10,3,1,val); // temp has to be divided by 10
    break;
    case '7' :
          mactopic = mactopic + "/" + "lux";
          dtostrf(value,0,0,val);
     break;
    case '8' :
          mactopic = mactopic + "/" + "hum";
          dtostrf(value,0,0,val);
     break;
    default:
    trc("can't read values");
    return false;
    }
    client.publish((char *)mactopic.c_str(),val);;
    trc(String(val));
    return true;
  }

#endif
```

betonishard, 2018-03-11:
@1technophile I changed the scan duration to 20. It stopped working after 2 hours this time.

betonishard, 2018-03-11:
@1technophile set duration to 20, TimeBtw_Read to at least 50000. Let you know what happens!
Thanks!

betonishard, 2018-03-14:
@1technophile again 4 days.. just quit 2 hours ago

walthowd, 2018-03-14:
Running three days now, but I'm being bad and running with a scan duration of 2 and TimeBtw_Read of 05000. I will report in again if mine makes it past four days.

1technophile, 2018-03-14:
I'm testing a version that doesn't use the multi core functionnality, still running since 3 days let's see how it goes.

walthowd, 2018-03-16:
My ESP32 quit right around four days as well.

1technophile, 2018-03-16:
If I sum up:
-ESP32 BLE gateway using multicore functionnality stop after 4 days
-ESP32 BLE gateway without multicore functionnality lasted more than 5 days (my PC rebooted due to a windows update...)

At this time I propose to remove the multicore functionnality so as to enable people who wants BLE only to run it correctly. Knowing that if RF is added, RF gateway will not be able to listen RF signal when doing a BLE scan. 
In the meantime I continue investigating on multicore.

What do you think?

betonishard, 2018-03-17:
@1technophile Not ideal, but I think this is a workaround that would be sufficient. 
Thanks for investigating!

1technophile, 2018-03-17:
Not yet I need to push a modification. In the meantime you can copy paste the modifications suggered above

betonishard, 2018-03-22:
@1technophile no luck, it still broke down after a couple of days...

[31 more replies]

closed 2018-05-24 by 1technophile