1technophile/OpenMQTTGateway #175: Home Assistant can't read sensors and some troubles on RF tx

LucaAghemo, 2018-02-15:
Hello, and thx for the great work you did with OpenMQTTGateway!
I build a combined version with RF rx and tx, IR rx and tx and a DHT22 sensor on a NodeMCU ESP8266 12E. The OMG version is 0.6.2.
RF and IR seems to work fine, I still have to figure out how to handle IR in Home Assistant, but this will come later.
RF has something strange: I used superetherodyne rx and tx, the same that I was using before connected on my RPI3. The weird thing is that on the RPI I used to control several devices but, with the same codes, on the 8266 it just work for certain devices, and others do not work anymore. 
I think it might be related to Pulse value, could you tell me how to set it? On the RPI 182 worked for all, I don't know what's the default setting in OMG.
The binary sensor in HA that should tell if the 8266 is connected also doesn't work, it reads "disconnected" even if it is actually up and working.
Sorry for the multiple questions, and sorry if I asked noob questions, that's what I am btw :-)
Thx in advance.

1technophile, 2018-02-15:
Hello,

Did you checked the wiki for RF advanced parameters sending?
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-RF

About home assistant integration you have also some infos in the wiki:
https://github.com/1technophile/OpenMQTTGateway/wiki/Home-Assistant-integration

Moreover for questions I advise you to go to the chat room:
[image]

LucaAghemo, 2018-02-16:
Thx a lot, in my fumbling around in search of info I read the wiki when I wasn't ready to understand it! 
Actually the Home Assistant integration page gives more info than what I remember, and gives me the chance to make some test and sort it out. If I'll be in trouble I'll go to the chat room. 
Anyway, you did an amazing work with this OMG, so thanks again!

LucaAghemo, 2018-02-16:
A small update, I solved the RF switch problem by changing the PLSL_xxx value to the correct one, that for my outlets was 345. Now it works perfectly!
I'll check on the chat for the binary sensor not working and other stuff.
THX!!

1technophile, 2018-02-22:
I think we can close this issue, if the problem remains reopen it

closed 2018-02-22 by 1technophile