1technophile/OpenMQTTGateway #178: Translating the codes

Rudertier, 2018-02-17:
Hi,
I am having  problems with receiving signals from the receiver, however I know that my transmitter and the nodemcu board I have can switch my plug, as I can run the send example out of the box. 
So I know that those commands work:
`
mySwitch.switchOn("11111", "00010");
mySwitch.switchOff("11111", "00010");
mySwitch.send(5393, 24);
mySwitch.send(5396, 24);
mySwitch.send("000000000001010100010001");
mySwitch.send("000000000001010100010100");
mySwitch.sendTriState("00000FFF0F0F");
mySwitch.sendTriState("00000FFF0FF0");
`
Now how does this translate to an MQTT payload? As I tried to say, I can use the default from SendDemo.ino.

I hope you can understand my question. 
Best and thanks

1technophile, 2018-02-17:
Hello,

From what I understand you don't understand how to send rf signal with the gateway ? 
Did you checked the user guide : 
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-RF#send-data-by-mqtt-with-advanced-rf-parameters

The first example could help you

Nevertheless this kind of question should be asked on the gitter chat

Rudertier, 2018-02-18:
Thanks for the prompt reply. Perhaps I did not phrase my question well enough.
As far as I understand I need to know three parameters: Pulslength, rf protocol and bit lenght. 
Now as I know that `mySwitch.switchOn("11111", "00010")` switches my plug when I run the *SendDemo.ino* I thought It could work my way backwards from there to determine the tree parameters I have to use. I would guess they are the default parameters of RCSwitch.

Thanks again for helping and I'll check out gitter.

1technophile, 2018-02-18:
to know what we have to send the best way is to check what the gateway or a ReceiveDemo_Advanced example are recceiving and give it to us. We will explain you how to translate it into MQTT the equivalent to:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-RF#send-data-by-mqtt-with-advanced-rf-parameters

Rudertier, 2018-02-19:
So there is no way, at least not an easy one, to get this based on the system and unit codes? That is a pity. I'll try to get one my receivers working then. Thanks for the help.

closed 2018-02-19 by Rudertier

1technophile, 2018-02-19:
It is not implemented, the way it is handled use the value from:
mySwitch.send(5393, 24);