1technophile/OpenMQTTGateway #181: ESP32 error: Sketch too big

ghianciulu, 2018-03-06:
hi, i'm trying to compile the sketch for esp32 dev kit this is the result:

Sketch uses 1452958 bytes (110%) of program storage space. Maximum is 1310720 bytes.
Global variables use 99572 bytes (33%) of dynamic memory, leaving 195340 bytes for local variables. Maximum is 294912 bytes.

Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board ESP32 Dev Module.

1technophile, 2018-03-06:
Hi Could you post your user_cobfig.h please without credentials

1technophile, 2018-03-07:
This is due to the last changes on ESP32 env, if you want to reduce the sketch size you can use this revision:
https://github.com/espressif/arduino-esp32/archive/a153f6ad280e5932be9c80341556348ff505d586.zip

1technophile, 2018-03-07:
https://github.com/espressif/arduino-esp32/issues/1075

walthowd, 2018-03-09:
Thanks, this got me up and going today.

closed 2018-03-23 by 1technophile