1technophile/OpenMQTTGateway #183: Can't read data_length

joeschny, 2018-03-08:
- OpenMQTTGateway version used ( Master )
- Firmware solution of Xiaomi device 2.7
- Using Wemos Lolin32 lite jointly with the 'Flower Care' Device from Xiaomi,

Log states that "Processing mi flora data - can't read data_length". Would appreciate to understand why?

Tried to understand well documented code but could not understand how/wher the specific handle for the temp getting is used. I read that a specific service/character UUID must be requested to receive data and i don't find it. 
Where to find the request in the code to get that data for that characteristic?

**Log**

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (S⸮L⸮⸮Ғ⸮Ҫ⸮j

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Connecting to 
(removed)
.
.
.
.
.
.
.
WiFi ok with manual config credentials
OpenMQTTGateway mac: 
B4:E6:2D:89:14:8D
OpenMQTTGateway ip: 
192.168.15.93
1883
Connecting to MQTT by IP adress
192.168.15.199
ZgatewayBT ESP32 setup done 
MQTT connection...
BT Task running on core 0
Connected to broker
Subscription OK to the subjects
Get Name 
LE-Bose Revolve SoundLink
Get ManufacturerData 
@0
Get RSSI 
home/BTtoMQTT/0452C7FB932A -74
Get TXPower 
10
Get ManufacturerData 
L
Get RSSI 
home/BTtoMQTT/A0EDCDDB8103 -58
Get ManufacturerData 

Get RSSI 
home/BTtoMQTT/32B5C4D990AE -91
Get ManufacturerData 
L
Get RSSI 
home/BTtoMQTT/18EE6927025B -65
Get ManufacturerData 
L
Get RSSI 
home/BTtoMQTT/1C1AC07C6B15 -88
Get ManufacturerData 
L
Get RSSI 
home/BTtoMQTT/1499E2138791 -82
Get ManufacturerData 
L
Get RSSI 
home/BTtoMQTT/18620B69552D -94
Get ManufacturerData 
L
Get RSSI 
home/BTtoMQTT/44B06F6D066F -89
Get ManufacturerData 
u
Get RSSI 
home/BTtoMQTT/F8042E86B5CB -86
Get ManufacturerData 
L
Get RSSI 
home/BTtoMQTT/6C448A0E6218 -95
Get Name 
Flower care
Get RSSI 
**home/BTtoMQTT/C47C8D661259 -76
Get service data 
310298003c5912668d7cc40d
Get service data UUID
0000fe95-0000-1000-8000-00805f9b34fb
Processing mi flora data
can't read data_length
⸮**
Get Name 
cc:04:b4:0a:71:d8
Get ManufacturerData 
SN⸮
Get RSSI 
home/BTtoMQTT/CC04B40A71D8 -86
Remaining memory
53332
E (35755) BT: btc_search_callback  BLE observe complete. Num Resp 166

BT Task running on core 0
....

joeschny, 2018-03-08:
Hold off reviewing ... just updated firmware to newest version and it seems to work for now...
still evaluating.

joeschny, 2018-03-08:
.. and it does not work. here the log. Temperature, lux and hum jump.

Log:
E (35790) BT: btc_search_callbaceׁJun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3⸮⸮⸮⸮⸮len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Connecting to 
removed
.
.
.
.
.
.
.
WiFi ok with manual config credentials
OpenMQTTGateway mac: 
B4:E6:2D:89:14:8D
OpenMQTTGateway ip: 
192.168.15.93
1883
Connecting to MQTT by IP adress
192.168.15.199
ZgatewayBT ESP32 setup done 
BT Task running on core 0MQTT connection...

Connected to broker
Subscription OK to the subjects
-->Filter for flora
0000fe95-0000-1000-8000-00805f9b34fb
found Flora
Get Name 
Flower care
Get RSSI 
home/BTtoMQTT/C47C8D661259 -69
Get service data 
712098005d5912668d7cc40d071003110000
Get service data UUID
0000fe95-0000-1000-8000-00805f9b34fb
Processing mi flora data
7
17.00
17
Remaining memory
53000
E (35771) BT: btc_search_callback  BLE observe complete. Num Resp 186

BT Task running on core 0
-->Filter for flora
0000fe95-0000-1000-8000-00805f9b34fb
found Flora
Get Name 
Flower care
Get RSSI 
home/BTtoMQTT/C47C8D661259 -68
Get service data 
71209800605912668d7cc40d041002d400
Get service data UUID
0000fe95-0000-1000-8000-00805f9b34fb
Processing mi flora data
5
212.00
21.2
Remaining memory
52976

1technophile, 2018-03-08:
hmm, what does not work, the esp32 is able to read the mi flora data. You mean that you don't see the value on your broker?

joeschny, 2018-03-09:
Apologies, for being so unspecific and thanks for your effort.

I understand now that the Temperature, Humidity and Lux information comes as part of the BLE advertising.  A client doesn't have to connect to the server and request separately the information.
The information comes with the additional ServiceData.

This information is now being provided to MQTT and i am ok so far.
i have one question left. When the advertising info for that service is received why are not all three values at once update? From what i gather every time only one valid information (Temperature, Humidity and Lux) is being provided it seems - is this correct? My thought was i get the service data and update all three items and send three mitt messages.

Thank you again

1technophile, 2018-03-10:
Yes you are correct one information is provided per scan.
Service data gives each round either temperature, lux, humidity or fertility

joeschny, 2018-03-10:
Uff, great, thank you. I got it.

closed 2018-03-10 by joeschny