1technophile/OpenMQTTGateway #184: ESP32: reboot loop

oakbrad, 2018-03-13:
I am trying to get this going on an ESP32 but it is stuck in a reboot loop.

From the serial monitor:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Connecting to 
bradnet
.
.
WiFi ok with manual config credentials
OpenMQTTGateway mac: 
30:AE:A4:44:AA:84
OpenMQTTGateway ip: 
192.168.0.112
1883
Connecting to MQTT by IP adress
192.168.0.6
ZgatewayBT ESP32 setup done 
BT Task running on core 0
abort() was called at PC 0x4016b483 on core 1

Backtrace: 0x4008af54:0x3ffda4f0 0x4008b053:0x3ffda510 0x4016b483:0x3ffda530 0x4016b4ca:0x3ffda550 0x4016b577:0x3ffda570 0x4016b5fa:0x3ffda590 0x400d52dd:0x3ffda5b0 0x400d5358:0x3ffda5d0 0x400d5377:0x3ffda5f0 0x400d3113:0x3ffda620 0x401851e2:0x3ffda650

Rebooting...
ets Jun  8 2016 00:22:57
```

I am using the ESP32 dev environment as mentioned in #181 

I have commented out the RF Gateway in my user_config.h, so just BT enabled

1technophile, 2018-03-14:
Hello,

Please provide more details:
-OMG version
-ESP32 hardware board
-parameters used to flash

Is your ESP32 standalone?

It is more likely that the board doesn't succeed in connecting to the MQTT server, please check your mqtt parameters and credentials.

oakbrad, 2018-03-14:
Well, I have reflashed with a fresh copy of OMG 0.6.2 and now it boots and connects to MQTT. 

But I get no BLE devices discovered, from serial monitor:

```
BT Task running on core 0
E (154031) BT: btc_search_callback  BLE observe complete. Num Resp 0

BT Task running on core 0
E (184235) BT: btc_search_callback  BLE observe complete. Num Resp 0

BT Task running on core 0
E (214439) BT: btc_search_callback  BLE observe complete. Num Resp 0

BT Task running on core 0
E (244643) BT: btc_search_callback  BLE observe complete. Num Resp 0

BT Task running on core 0
```

I have ESP32 from HiLetGo https://www.amazon.com/gp/product/B0718T232Z/

Flashed in Arduino IDE with "ESP32 Dev Module" and have left the settings at their default

Anything I can do to troubleshoot? Thank you

walthowd, 2018-03-14:
You are going to want OMG version 0.7 from the latest GitHub master branch as it has WiFi reconnect logic in it. 

When I first flashed 0.62 the BLE detection wasn't finding any devices either. If I kicked it off WiFi it always found BLE devices and then after a few more boots it seemed to work each time.

oakbrad, 2018-03-15:
Ah, I see what I did. I just reflashed with the latest 0.7 and now I am back in the reboot loop

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Connecting to 
bradnet
.
.
WiFi ok with manual config credentials
OpenMQTTGateway mac: 
30:AE:A4:44:AA:84
OpenMQTTGateway ip: 
192.168.0.114
1883
Connecting to MQTT by IP adress
192.168.0.6
ZgatewayBT ESP32 setup done 
BT Task running on core 0
abort() was called at PC 0x4016b47b on core 1

Backtrace: 0x4008af54:0x3ffda4f0 0x4008b053:0x3ffda510 0x4016b47b:0x3ffda530 0x4016b4c2:0x3ffda550 0x4016b56f:0x3ffda570 0x4016b5f2:0x3ffda590 0x400d52d5:0x3ffda5b0 0x400d5350:0x3ffda5d0 0x400d536f:0x3ffda5f0 0x400d310b:0x3ffda620 0x401851da:0x3ffda650

Rebooting...
ets Jun  8 2016 00:22:57
```

I am not seeing it connect to MQTT, although my settings are the same as before. My MQTT broker does not require a password - I just have = "" for the user and password. Same settings worked with 0.6.2, I would at least see it connect and publish a LWT

walthowd, 2018-03-15:
Did you modify the MQTT max packet size I'm pubsub? Just a shot in the dark, because I believe you are crashing where it should be subscribing to its own topics.

oakbrad, 2018-03-15:
I have it set to 512 in pubsubclient.h

Thank you it does look like it's crashing when connecting to MQTT, I am not seeing it connect and announce on my broker.

1technophile, 2018-03-15:
I think the crash is due to line 512 of OpenMQTTGateway.ino:
      if(WiFi.getAutoReconnect()) WiFi.onEvent(WiFiEvent);

try to comment it and you will see that the root issue is that OMG doesn't succeed in connecting to your broker

oakbrad, 2018-03-15:
@1technophile thank you, actually now with that commented out it does successfully connect to my broker :)

```WiFi ok with manual config credentials
OpenMQTTGateway mac: 
30:AE:A4:44:AA:84
OpenMQTTGateway ip: 
192.168.0.114
1883
Connecting to MQTT by IP adress
192.168.0.6
ZgatewayBT ESP32 setup done 
BT Task running on core 0
MQTT connection...
Connected to broker
Subscription OK to the subjects
Remaining memory
59256
E (33212) BT: btc_search_callback  BLE observe complete. Num Resp 0

BT Task running on core 0
Remaining memory
58920
E (63416) BT: btc_search_callback  BLE observe complete. Num Resp 0

BT Task running on core 0
```

And it does successfully publish it's LWT this time
```
home/OpenMQTTGateway/LWT Online
home/OpenMQTTGateway/version 0.7
```

So 0.6.2 and 0.7.0 (with line 512 commented out) seem to have same behavior - no BLE devices detected.

1technophile, 2018-03-15:
are you sure you have BLE devices emitting?

oakbrad, 2018-03-15:
I've got about a dozen MiFlora devices and 4 iBeacons within range, I can see them if I scan with my phone or computer.

1technophile, 2018-03-15:
try to:
**In OpenMQTTGateway.ino replace in line 589:**
```
    #ifdef ZgatewayBT
      #ifndef ESP32
        if(BTtoMQTT())
        trc(F("BTtoMQTT OK"));
      #endif
    #endif
```
by 

```
    #ifdef ZgatewayBT
        if(BTtoMQTT())
        trc(F("BTtoMQTT OK"));
    #endif
```

**and replace all the content of ZgatewayBT by this one:**
```
/*  
  OpenMQTTGateway  - ESP8266 or Arduino program for home automation 

   Act as a wifi or ethernet gateway between your 433mhz/infrared IR signal/BLE  and a MQTT broker 
   Send and receiving command by MQTT
 
  This gateway enables to:
 - publish MQTT data to a different topic related to BLE devices rssi signal

    Copyright: (c)Florian ROBERT
  
    This file is part of OpenMQTTGateway.
    
    OpenMQTTGateway is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    OpenMQTTGateway is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.

Thanks to wolass https://github.com/wolass for suggesting me HM 10 and dinosd https://github.com/dinosd/BLE_PROXIMITY for inspiring me how to implement the gateway
*/
#ifdef ZgatewayBT

  #ifdef ESP32
    /*
       Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
       Ported to Arduino ESP32 by Evandro Copercini
    */
    // core task implementation thanks to https://techtutorialsx.com/2017/05/09/esp32-running-code-on-a-specific-core/
    
    #include <BLEDevice.h>
    #include <BLEUtils.h>
    #include <BLEScan.h>
    #include <BLEAdvertisedDevice.h>

    //Time used to wait for an interval before resending BLE infos
    unsigned long timeBLE= 0;
      
    class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
          void onResult(BLEAdvertisedDevice advertisedDevice) {
            String mac_adress = advertisedDevice.getAddress().toString().c_str();
            mac_adress.replace(":","");
[... 261 lines cut ...]
    break;
    case '4' :
          mactopic = mactopic + "/" + "tem";
          if (value > 65000) value = value - 65535;
          dtostrf(value/10,3,1,val); // temp has to be divided by 10
    break;
    case '7' :
          mactopic = mactopic + "/" + "lux";
          dtostrf(value,0,0,val);
     break;
    case '8' :
          mactopic = mactopic + "/" + "hum";
          dtostrf(value,0,0,val);
     break;
    default:
    trc("can't read values");
    return false;
    }
    client.publish((char *)mactopic.c_str(),val);;
    trc(String(val));
    return true;
  }

#endif
```

oakbrad, 2018-03-15:
That did it, I now see my devices

Any idea why no MiFlora data?

In serial monitor
```
Get Name 
Flower mate
Get RSSI 
home/BTtoMQTT/C47C8D60E511 -91
Get service data 
310298000011e5608d7cc40d
Get service data UUID
0000fe95-0000-1000-8000-00805f9b34fb
Processing mi flora data
can't read data_length
Get Name 
Flower care
Get RSSI 
home/BTtoMQTT/C47C8D633A4D -89
Get service data 
310298001b4d3a638d7cc40d
Get service data UUID
0000fe95-0000-1000-8000-00805f9b34fb
Processing mi flora data
can't read data_length
```
Published to MQTT
```
home/BTtoMQTT/A01828E95B35/ManufacturerData L
home/BTtoMQTT/A01828E95B35 -77
home/BTtoMQTT/EC358651DC5B/ManufacturerData L
home/BTtoMQTT/EC358651DC5B -52
home/BTtoMQTT/65E98446EF16/ManufacturerData L
home/BTtoMQTT/65E98446EF16 -52
home/BTtoMQTT/C47C8D662167/name Flower care
home/BTtoMQTT/C47C8D662167 -84
home/BTtoMQTT/C47C8D627168/name Flower mate
home/BTtoMQTT/C47C8D627168 -71
home/BTtoMQTT/C47C8D627B7D/name Flower mate
home/BTtoMQTT/C47C8D627B7D -92
home/BTtoMQTT/E0E5CFAF63ED/name iBeacon-8100
home/BTtoMQTT/E0E5CFAF63ED/ManufacturerData L
home/BTtoMQTT/E0E5CFAF63ED -84
home/BTtoMQTT/E0E5CFAF6E9E/name iBeacon-8082
home/BTtoMQTT/E0E5CFAF6E9E/ManufacturerData L
home/BTtoMQTT/E0E5CFAF6E9E -85
home/BTtoMQTT/7AC6D8D069C4/ManufacturerData �
home/BTtoMQTT/7AC6D8D069C4 -90
home/BTtoMQTT/C47C8D62785B/name Flower mate
home/BTtoMQTT/C47C8D62785B -81
home/BTtoMQTT/C47C8D660ED0/name Flower care
home/BTtoMQTT/C47C8D660ED0 -86
home/BTtoMQTT/5CF82183F96B/name Tz14485
home/BTtoMQTT/5CF82183F96B/ManufacturerData L
home/BTtoMQTT/5CF82183F96B -86
```

1technophile, 2018-03-15:
did you tried to update the mi flora internal firmware?

oakbrad, 2018-03-15:
Will try that, thanks for your help

closed 2018-03-15 by oakbrad