1technophile/OpenMQTTGateway #185: Missing wiki entry for Mi flora BT LE integraton

AKA-47, 2018-03-16:
Hi. I try to get mi flora running with openmqtt gateway on esp32.

the only thing i see pn mqtt is 
home/BTtoMQTT/C47xxxxx4C2F/name Flower care
home/BTtoMQTT/C47xxxxx4C2F -51

so the gate sees the miflora device but nothing else.
Do i have to pair it somehow? Shouldn't we create a wiki about this?
Pairing within the official app seems not to work...

I'm using version 0.6.2master28022018 (yes i saw the fix "reinit BLE each time we launch a scan to avoid BLE stop" and will recompile later)

any ideas?

1technophile, 2018-03-16:
Hello,

did you try to update your mi flora device

1technophile, 2018-03-16:
Could you post the log yopu are getting?

AKA-47, 2018-03-16:
...no logs... the only thing i can say with some battery in-out procedure : there is firmware 2.7.0 which want to be upgraded to 3.1.8.. i will try to update somehow...

AKA-47, 2018-03-18:
I am still unable to connect.. do i really need to connect the app? Maybe the SeviceData is just different from the one you found out...
I contacted the reseller and he say that he sent the international version... I'm still not sure about this....
I will now again try other devices to maybe get the device updated or are there any other ideas?

AKA-47, 2018-03-21:
..ok now i have FW Version 3.1.8 on my device....it works with OpenMQTT Gateway version 0.6.2 but not higher because of some "Multicore" Problems .. see https://github.com/1technophile/OpenMQTTGateway/issues/186

We should create a Wili entry showing how to use ist:

MQTT Readings:
home/BTtoMQTT/C47xxxxxxx2F/name Flower care
home/BTtoMQTT/C47xxxxxxx2F/fer 56
home/BTtoMQTT/C47xxxxxxx2F/tem 20.5
home/BTtoMQTT/C47xxxxxxx2F/lux 69
home/BTtoMQTT/C47xxxxxxx2F/hum 18

And VERY Important: TimeBtw_Read to configure the time between readings!

1technophile, 2018-03-24:
yes don't hesitate to update the wiki so as to enable others to set it uo easily

1technophile, 2018-04-04:
added into the wiki, don't hesitate to amend it if necessary

closed 2018-04-04 by 1technophile

AKA-47, 2018-04-05:
I found out that we could also read battery stats by raspberry pi. Any plan to implement this on esp?

1technophile, 2018-04-05:
I propose you to open an issue requesting for that, if someone wants to help implementing it, he will take the idea. In my side I will focus in the next months on code refactoring and other gateways implementations (lora, gsm)

antibill51, 2018-04-12:
Hi. Is there an example of mi flora integration in home assistant with openmqttgateway? 
Thanks

AKA-47, 2018-04-12:
I can only give you an openhab 2 example