1technophile/OpenMQTTGateway #188: Add SHT31 and E-Paper support

knopserl, 2018-03-24:
I was thinking to use this project for a quality temperature/humidity sensor with e-paper (Wavershare, 1,54" 200x200 pixel) with LiPo battery.
I would need some help to add support for a SHT31 temp/humitidy sensor and also for a waveshare (or any other) e-paper/e-ink. The libs are available but if somebody could proper integrate it into the project, that would be great.

BertrumUK, 2018-03-27:
Could this be added as SHT3x to include other versions of this sensor?

1technophile, 2018-03-27:
yes of course ;-)

1technophile, 2018-07-18:
great, could you submit a pull request so as to enable me to add it easily?

1technophile, 2018-07-28:
@stripeyhorse do you need some help so as to submit a pull request?

stripeyhorse, 2018-07-31:
@1technophile Yes please. i did try and do it but ive ended up with https://github.com/stripeyhorse/OpenMQTTGateway

My SHT3x sensor on my m5 stack has been working great all week. I think my CCS811 is faulty thou so ive stopping using it. I have ordered more and a few different sensors as well. The code is pretty hacked together but i did try and follow the layout and style already used.

1technophile, 2018-07-31:
@stripeyhorse 
When on your [OMG repository main page](https://github.com/stripeyhorse/OpenMQTTGateway) you should see a button "New pull request".
Click on it
You will see the changes you have made.
Check these changes.
If ok click on "create pull request"
Name it and put some descriptions

Once done I will review the changes made

insane2subro, 2018-08-01:
In the mean time how about adding support for SSD1306 oled displayes, these are pretty common.

1technophile, 2018-09-07:
@insane2subro feel free to submit a pull request also :-)