1technophile/OpenMQTTGateway #190: Issues with RFM69/Ethernet shield

vador98, 2018-04-01:
Hi all,

I've been working on getting this Gateway to run on my Uno with an ethernet sheild and an RFM69HW chip and I've been running into some problems.  I've gotten to the point where the ethernet will successfully connect to DHCP, the RFM chip will successfully initialize, and the MQTT server is successfully connected.  However, despite seeing all success notifications in the serial prints, It's not receiving any RF signals.  I've tried loading the "Gateway" example code, and everything is received successfully.  As soon as I load the OpenMQTTGateway, nothing is received.  This tells me that everything is wired successfully, and that my RFM69 headers are properly configured.

Any suggestions as to what I could be doing wrong?  I can submit whatever parts of my code you feel are helpful...  I just don't have a clue as to what could be wrong. 

I'm running OpenMQTTGateway v0.6.2
RFM69 - Current version (I'm unable to find a version, but this was pulled on March 29th or so)
RCSwitch - v2.6.2

1technophile, 2018-04-01:
Hello,

What is your sender  ?  Is it rfm69 based ? 
Rfm69 librarie used currently support only receiving frpm other rfm69.
Rcswitch is not used in rfm69 gateway.

vador98, 2018-04-01:
The sending device is an anarduino with an rfm69 chip. When using the basic example gateway, I do receive signals from it.  I’m only using RC Switch because the code seemed to require it, but that may have been because I had OMG misconfigured initially.

1technophile, 2018-04-01:
To avoid the need of rcswitch, just comment ZgatewayRF

Here is the send example I use on an esp to test the gateway:

```
/* RFM69 library and code by Felix Rusu - felix@lowpowerlab.com
// Get libraries at: https://github.com/LowPowerLab/
// Make sure you adjust the settings in the configuration section below !!!
// **********************************************************************************
// Copyright Felix Rusu, LowPowerLab.com
// Library and code by Felix Rusu - felix@lowpowerlab.com
// **********************************************************************************
// License
// **********************************************************************************
// This program is free software; you can redistribute it 
// and/or modify it under the terms of the GNU General    
// Public License as published by the Free Software       
// Foundation; either version 3 of the License, or        
// (at your option) any later version.                    
//                                                        
// This program is distributed in the hope that it will   
// be useful, but WITHOUT ANY WARRANTY; without even the  
// implied warranty of MERCHANTABILITY or FITNESS FOR A   
// PARTICULAR PURPOSE. See the GNU General Public        
// License for more details.                              
//                                                        
// You should have received a copy of the GNU General    
// Public License along with this program.
// If not, see <http://www.gnu.org/licenses></http:>.
//                                                        
// Licence can be viewed at                               
// http://www.gnu.org/licenses/gpl-3.0.txt
//
// Please maintain this license information along with authorship
// and copyright notices in any redistribution of this code
// **********************************************************************************/

#include <RFM69.h>    //get it here: https://www.github.com/lowpowerlab/rfm69
#include <SPI.h>

//*********************************************************************************************
// *********** IMPORTANT SETTINGS - YOU MUST CHANGE/ONFIGURE TO FIT YOUR HARDWARE *************
//*********************************************************************************************
#define NETWORKID     200  // The same on all nodes that talk to each other
#define NODEID        1  // The unique identifier of this node
#define RECEIVER      10    // The recipient of packets

//Match frequency to the hardware version of the radio on your Feather
#define FREQUENCY     RF69_433MHZ
//#define FREQUENCY     RF69_868MHZ
//#define FREQUENCY     RF69_915MHZ
#define ENCRYPTKEY    "sampleEncryptKey" //exactly the same 16 characters/bytes on all nodes!
#define IS_RFM69HCW   true // set to 'true' if you are using an RFM69HCW module

//*********************************************************************************************
#define SERIAL_BAUD   115200

#if defined(__AVR_ATmega168__) || defined(__AVR_ATmega328P__) || defined(__AVR_ATmega88) || defined(__AVR_ATmega8__) || defined(__AVR_ATmega88__)
#define RFM69_CS      10
#define RFM69_IRQ     2
#define RFM69_IRQN    digitalPinToInterrupt(RFM69_IRQ)
#define RFM69_RST     9
#define LED           13  // onboard blinky
#elif defined(__arm__)//Use pin 10 or any pin you want
// Arduino Zero works
#define RFM69_CS      10
#define RFM69_IRQ     5
#define RFM69_IRQN    digitalPinToInterrupt(RFM69_IRQ)
#define RFM69_RST     6
#define LED           13  // onboard blinky
[... 67 lines cut ...]
  loops = 10;
  startMillis = millis();
  while (loops--) {
    if (radio.sendWithRetry(RECEIVER, radioPerf, strlen(radioPerf)+1)) {
      deltaMillis = millis() - startMillis;
      Serial.print(" OK ");
      Serial.println(deltaMillis);
      break;
    }
    else {
      Serial.print("!");
    }
    delay(50);
  }
  if (loops <= 0) {
    Serial.println(" Fail");
    deltaMillis = 0;
  }

  radio.receiveDone(); //put radio in RX mode
}
```
You can check the differences with yours.

Regarding the gateway do you see something on your MQTT broker?

vador98, 2018-04-01:
I tried your transmitter code and it's not working either.  Hopefully I'll have more time to devote to this tomorrow.  

In regards to MQTT, I'm not certain that that piece is completely working.  It's connecting to the broker, but I don't think it's subscribing to any topics.  I've not had a chance to diagnose that piece yet.  More to come.

Thank you for your help!

vador98, 2018-04-02:
With promiscuous mode on, I seem to occasionally receive 2 random messages, which are always the same.  They are successfully being published to MQTT, but they are not readable.  Here are the latest 2 messages received: 
[image]

1technophile, 2018-04-02:
Could you try with the library version that I have tested with:
https://github.com/LowPowerLab/RFM69/tree/928de01a2b0ffc21d0dff935694641c04d6b9041

vador98, 2018-04-02:
Trying with that library, there's no difference.  I've attached my code, in hopes it will help. 

Also, here's how I have everything wired:
I believe that the W5100 shield uses the following:
CS - 10
MOSI - 11
MISO - 12
SCK - 13

RFM69:
CS - 7
MOSI - 11
MISO - 12
SCK - 13

In OpenMQTTGateway.ino, I've set the pinmode of pins 7 and 4 and set them to HIGH to disable the RFM69 and SD Slot during Ethernet init.  Without this, the ethernet wasn't able to pull an IP via DHCP.   I've tried setting pin 7 to LOW after ethernet init, but saw no improvement.  
[OpenMQTTGateway.zip](https://github.com/1technophile/OpenMQTTGateway/files/1868109/OpenMQTTGateway.zip)

vador98, 2018-04-02:
I've noticed that as I leave the monitor open for extended periods of time, no additional messages are received - I only see the duplicated RFM data immediately after the arduino is restarted.  Do you think that the code is freezing up somewhere?

vador98, 2018-04-03:
I think I've made progress!  I've discovered that the code runs until the first message is received.  At that point, the code freezes.  I'm afraid I likely won't have time to look at this again for a few hours, but I'll post any other updates as I find them.

vador98, 2018-04-03:
I've found the following gateway code that does consistently receive messages:
https://github.com/abouillot/HomeAutomation

Due to the way it's coded, the messages aren't in the right format for massing on to MQTT, but that's a short term problem.  I'm not sure which direction is better worth my time - trying to fix this code using the HomeAutomation code, or changing the HomeAutomation code to work with my nodes...

Since I'm not planning on using other gateway items (IR, BT, etc), perhaps switching to the other code may be the better way to go since I can follow it easier.

1technophile, 2018-04-03:
Hello,

I will try to reproduce your issue in the following days and I will give you some feedback, thanks for the analysis in all the case

fluppie, 2018-08-16:
I'm using the latest RFM69 library with two mods as specified here: https://github.com/boaschti/MQTT_WLan_RFM69_Gateway

> // Required Modifications:
> // 1. PubSubClient.h #define MQTT_MAX_PACKET_SIZE 256
> // 2. add delay(1); in line 244 in RFM69.cpp to reset watchdog
> 

With that gateway I receive packets but it's not in the form that I want. So I tried your Gateway, keeping the mods from above.

> mounted file system
reading config file
opened config file
{"mqtt_server":"192.168.1.29","mqtt_port":"1883","mqtt_user":"emonpi","mqtt_pass":"emonpimqtt2016"}
parsed json
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.48
connected...yeey :)
saving config
{"mqtt_server":"192.168.1.29","mqtt_port":"1883","mqtt_user":"emonpi","mqtt_pass":"emonpimqtt2016"}*WM: freeing allocated params!
OpenMQTTGateway mac: 
5C:CF:7F:11:5D:69
OpenMQTTGateway ip: 
805415104
1883
Connecting to MQTT by IP adress
192.168.1.29
ZgatewayRFM69 initialization failed
ZgatewayRFM69 Listening and transmitting at
868
MQTT connection...
Connected to broker
Subscription OK to the subjects

> 

As you see it says initialisation failed. Any idea why that is?

wolass, 2018-12-06:
I am experiencing the similar issue withe the reversed ???? Signs.... I think it is hardware based. I have this issue in one more but the same configuration of rfm69hcw + Arduino pro mini + same sketch gave successful communication with OMG

closed 2019-08-14 by stale[bot]