1technophile/OpenMQTTGateway #191: MiFlora only send one value

asexauer, 2018-04-09:
I'm using ESP32 and a MiFlora device. But every time only one value either lux or fer is send to the MQTT-Server. On the console I see the same value beeing processed. It seems there is a loop missing to process all mi flora data?

AKA-47, 2018-04-09:
...same here but works...

1technophile, 2018-04-09:
@asexauer Mi flora service data diffuse alternatively the different measurement. It's not a one shot sending.
How many times did you wait to assume there is only two measurement returned?

1technophile, 2018-04-13:
hello, 

some news about that?

asexauer, 2018-04-14:
Hi, sorry was busy with other things. I looked at the data for about 30 minutes and has never seen humidity and temperatures. It would be great when all parameters that got fetched from the flora mi device would be sent directly one after the other.

1technophile, 2018-04-14:
We use the data advertised by the miflora service data. Service data expose one parameter at a time.
Did you updated your miflora firmware with the app?

1technophile, 2018-04-21:
@asexauer some news about that?

asexauer, 2018-04-21:
I have one device with firmware 2.7.0 and one with v3.1.8, for both no firmware update is available. I'm not sure why they are on different versions. I bought them separately. Today I hd the time to try it again. I got the same results, and was not able to obtain at least every sort of data during 30 minutes. But because I need to run it from solar power, I switched to a another solution and am fetching now all data from all sensors at once and go to deep sleep after that.
Thank you for your help. Maybe I have time to try OpenMQTTGateway sometime more in depth.

closed 2018-04-21 by asexauer

1technophile, 2018-04-21:
@asexauer ok one tip could be to change the time between readings in config_BT.h
`TimeBtw_Read`
by putting it to a value like 23333