1technophile/OpenMQTTGateway #192: Add relay/relays support (can be use as wireless thermostat, wall plug, etc)

PetricaM, 2018-04-10:
Hi Florian,

Could you add support for one relay channel or multiple channels (might be a problem on ESP8266 due to small number of available pins but should be ok for ESP32 and Mega/Uno)? Basically it would have a command and a state topic. In function needs to set the pin high when payload is 1/on and low when payload is 0/off then publish the new state to the dedicated topic.

It could be used to power a wireless thermostat when connecting a relay to a NodeMCU. Also, combined with an ACS712 can provide for a wireless plug :)

One channel relay:
https://www.aliexpress.com/item/Free-Shipping-5V-low-level-trigger-One-1-Channel-Relay-Module-interface-Board-Shield-DC-AC/32801762863.html
Four channels:
https://www.aliexpress.com/item/Free-shipping-With-optocoupler-4-channel-4-channel-relay-control-panel-PLC-relay-5V-module-for/1885672929.html

PetricaM, 2018-04-13:
Hi Florian,

It works :) 

Could you also add acknowledgement  by publishing state of the relay to a state topic?

Thanks

1technophile, 2018-04-13:
Hi @PetricaM Yes of course, but I assume it is not anacknowledgement from a real state but only the fact that the gateway received and processed the instruction?

PetricaM, 2018-04-13:
Yes, of course. Although I think that the gateway's LWT state topic can be used as state for all the sensors (including the relay), there might be cases when the commands in the queue are not processed, right?

Otherwise we can put a current monitor sensor (obviously connected to another NodeMCU) on pin D2. And then another monitor sensor to monitor that the monitor sensor is monitoring what it needs to monitor :))

closed 2018-04-29 by 1technophile

PaulAntonDeen, 2019-10-31:
Am I correctly understanding that we can only actuate 1 relay this way? Would it be possible to actuate 2 relays by defining two output pins? Is there a way to get a bin download as I seem to have trouble compiling this in visual studio.

1technophile, 2019-10-31:
> Am I correctly understanding that we can only actuate 1 relay this way
Yes

>Would it be possible to actuate 2 relays by defining two output pins?
Some development is needed but a part of the work was done there:
https://github.com/1technophile/OpenMQTTGateway/pull/299/files
Unfortunately not finished

>  Is there a way to get a bin download as I seem to have trouble compiling this in visual studio
Yes on the release page
https://github.com/1technophile/OpenMQTTGateway/releases

PaulAntonDeen, 2019-11-01:
Sure, I'll test, what can I do?

PetricaM, 2019-11-01:
Haven't been able to compile OMG on Platformio (but didn't tried really hard to set it either :) )

On Arduino IDE I think that (Florian can correct me on this) you can do it quick and dirty by adding other ONOFF sections everywhere in User_config.h and Main.ino (ie. ZactuatorONOFF, ZactuatorONOFF2, ZactuatorONOFF3) and then do multiple copies of ZactuatorONOFF.ino and config_ONOFF.h files in the same folder.

However, don't simply do search and replace ONOFF with ONOFF2 or ONOFF3 all over the project as there are also sections specific to Sonoff.

1technophile, 2019-11-01:
> On Arduino IDE I think that (Florian can correct me on this) you can do it quick and dirty by adding other ONOFF sections everywhere in User_config.h and Main.ino (ie. ZactuatorONOFF, [...]

Yes you can do it like that indeed.

Nevertheless I will add a more flexible way of setting pins soom

1technophile, 2019-11-01:
You can test this branch:
https://github.com/1technophile/OpenMQTTGateway/tree/on-off-variable-pin

PaulAntonDeen, 2019-11-02:
Which lines to comment/uncomment from the platformio.ini (using NODEMCUv2)?
;default_envs = nodemcuv2-rf
;default_envs = rf-wifi-gateway
;default_envs = nodemcuv2-rf2
;default_envs = nodemcuv2-pilight
;default_envs = nodemcuv2-ir
;default_envs = nodemcuv2-ble
;default_envs = nodemcuv2-all

Anything else to configure before compiling?

1technophile, 2019-11-02:
answered here:
https://github.com/1technophile/OpenMQTTGateway/pull/481