1technophile/OpenMQTTGateway #196: Run on a Raspberry Pi?

flatsiedatsie, 2018-04-16:
I am currently using my Raspberry Pi as a [RF 433 cloner](https://github.com/flatsiedatsie/433Switches). It works really well. 

This software looks really nice though, so I was wondering: could it run on a Raspberry Pi?

1technophile, 2018-04-16:
not as it is, but what would the advan,tages on make it run to a pi?
I found that esp8266 or arduino have better stability, they are more "upload and don't take care about" compared to a pi which is a more complex platform and needs more maintenance.
Moreoever esp and arduino have a lower cost.

flatsiedatsie, 2018-04-21:
I'd like to have as few as possible wifi connected 2,4ghz ARM chips in my smart home, for security reasons.

1technophile, 2018-04-21:
hmmm interesting, could you precise your thoughts, I'm curious about these security reasons?

1technophile, 2018-05-04:
I'm closing this issue if you have more info don't hesitate to give them

jwillmer, 2018-05-18:
Maybe I can give you a use case:  I have 433Mhz radio communication in my flat and also Zigbee devices. To receive Zigbee I need a USB dongle. I found the [zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt) gateway that runs on raspberry pi but now I still need to receive 433Mhz communication. 
I have tested a couple of libraries that are available for raspberry pi and the one with the best results and performance is [pigpio](https://github.com/joan2937/pigpio). The problem is that this library still need 5-6% CPU since in queries all pins simultaneously. I haven't found a library that is flexible enough to just watch over 1 GPIO pin and have a great performance.

jwillmer, 2018-05-18:
Basically I could turn it around and ask for a Zigbee integration. I actually also think that simpler devices that only do one thing are much better then a complete linux that does only one thing.

1technophile, 2018-05-18:
yes maybe a feature request to add zigbee to OMG!

Indeed a pi to pass RF  signals without security layers like 433mhz base protocols is like having a bazooka to kill a mosquitto in my point of view. I'm not saying that it will not be usefull but just that at this time this is not the best approach in terms of costs/simplicity/reliability. 
I have an Arduino Uno running since more than 2 years with the first version of OpenMQTTGateway, and this is one of my most reliable Home automation component so far. Compared to my Pi that need regular update and had 3 major breakdowns with data loss ...

1technophile, 2018-06-27:
maybe a track to dig if someone is interested:
https://github.com/me-no-dev/RasPiArduino

1technophile, 2018-06-27:
another track is to check arduino create 
https://www.youtube.com/watch?v=nzx4x4OnBCE

1technophile, 2018-06-27:
I reopen the issue if some people wants to do the test on RasPiArduino or Arduino create

podarok, 2019-08-05:
Would be cool to run this in raspberry pi. I have disabled majority of linux servixes and it is reliable as hell...

happytm, 2019-08-08:
I have been using RPIEasy firmware on PI zero for a while and it is very mature with lot of common sensors already supported by it.It just had data logger added to it which is very useful for data collection and could be used with Node-Red or Home Assistant and MQTT broker,database,Home automation UI all reside on same PI Zero device.It make life simple and does not require cloud for local automation. I have just posted proposal for low power network. It will be great if Main Gateway is implemented to work with RPIEasy. RPIEasy already have serial interface. Links are below:

https://github.com/enesbcs/rpieasy

https://github.com/1technophile/OpenMQTTGateway/issues/421

Thanks

happytm, 2019-10-07:
Just wanted to let you know that RPIEasy mentioned above have already implemented ESPNow feature last week.

Thanks

closed 2019-12-14 by stale[bot]

poofyteddy, 2020-02-01:
I'll just drop in here because i'm also looking for a gateway who run on RPI and will allow me to republish BLE to MQTT.
The why rpi is because i already have rpiz all around the house doing TTS, Squeezelite, Wallscreen ... i really don't want to deploy more hardware for stuff already there.

happytm, 2020-02-02:
I Think best solution I found for your goal is https://github.com/enesbcs/rpieasy. Ihave been using it for 
months and it is very robust code.
Thanks.

flatsiedatsie, 2020-02-07:
To answer the original question: I'd like to run as few different devices as possible. If a Pi can manage 433 well enough (and my Domoticz plugin showed it was possible), then why not?

In the end, I did go the Arduino route:
https://www.candlesmarthome.com/signal-hub

1technophile, 2022-02-27:
For BLE you can now use Theengs Gateway on a raspberry pi or other hardware:
https://community.openmqttgateway.com/t/openmqttgateway-ble-on-windows-raspberry-pi-or-apple-mac/1824/2