1technophile/OpenMQTTGateway #197: Continuously send IR message

konektor001, 2018-04-18:
Is it possible to repeat X times an IR message other than Sony protocol?
It would be useful for example sending continuous "volume_up" to an amplifier.
Thanks.

1technophile, 2018-04-18:
yes just follow the wiki:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-IR#send-data-by-mqtt-with-advanced-ir-parameters

1technophile, 2018-04-18:
Indeed, i corrected it. It should work with sendNEC, try maybe a bigger repeat , 10 or 20

konektor001, 2018-04-18:
Sorry, but for me, the repeat command does not work. If I send quickly the single messages it works (not so effective for MQTT), but it would be better with the repeat command.

1technophile, 2018-04-22:
do you see ` IR repeat:` on the serial monitor?

1technophile, 2018-06-08:
I'm going to close the issue, you can reopen it if needed

closed 2018-06-08 by 1technophile