1technophile/OpenMQTTGateway #200: Dooes this firmware support SONOFF RF Bridge R2?

jmav, 2018-05-04:
I'we just received latest SONOFF RF Bridge R2 (https://www.youtube.com/watch?v=NscaZQpdedM). I did upload latest firmware and everything went well. I don't see any logs in console about wifi initialisation.  Dis any try this firmware on R2?
[image]

1technophile, 2018-05-04:
When you talk about the console are you talking about serial monitor or your mqtt console?

gabr777, 2018-06-06:
Hi, I have a the same issue.
Fw uploads from arduino IDE:
Uploading 382672 bytes from C:\Users\xxxxxx\AppData\Local\Temp\arduino_build_812836/OpenMQTTGateway.ino.bin to flash at 0x00000000
................................................................................ [ 21% ]
................................................................................ [ 42% ]
................................................................................ [ 64% ]
................................................................................ [ 85% ]
......................................................                           [ 100% ]
And serial console shows:
[image]

1technophile, 2018-06-08:
Did you use baudrate of 19200 in your serial monitor ?

starkillerOG, 2018-06-27:
@jmav  I recived my SONOFF RF Bridge R2 today, looks exactly like your picture.
I flashed OpenMQTTGateway on it and it worked right away for me, no problems at all.
I succesfully received signals from 8 diffrent buttons of a remote, 2 PIR sensors and a door remote.

I used the standard instructions: https://github.com/1technophile/OpenMQTTGateway/wiki/Sonoff-RFB
I used this picture: https://user-images.githubusercontent.com/804771/41503010-7b813536-71c8-11e8-9a54-bcf3af7a6c86.jpg from this github issue: https://github.com/arendst/Sonoff-Tasmota/issues/1916 to see which pins I needed to connect (I did not want to bend my LED on top).
(3.3V to 3.3V, RX to TX, TX to RX and GND to GND).

I used the latest OpenMQTTGateway software v0.7.
and I used the WifiManger setup, although I am going to flash it again tonight and try the manualwifisetup. With the WifiManger it won't reconect if I reboot my router because that takes to long and will go in Wifi setup mode. (I prefer to have it keep trying to connect until my router is back up and running).

Hope this info helps.

1technophile, 2018-06-27:
@starkillerOG With the last version of master and by extension in v0.8[ the issue ](https://github.com/1technophile/OpenMQTTGateway/pull/220)on where OMG is coming back to AP mode after a wifi lost is corrected

starkillerOG, 2018-06-27:
@1technophile Thank you for pointing that out, really usefull.
I have it all working now, and it reconnects fine after reboot of my router or MQTT Broker (tested it).

starkillerOG, 2018-06-28:
Why do you get messages on the topic SYStoMQTT?
which modules did you use?
You should only use the ZgatewaySRFB module, the rest schould be commented out in user_config.h:

```
/*-------------DEFINE THE MODULES YOU WANT BELOW----------------*/
//Addons and module management, comment the Z line

//#define ZgatewayRF          //ESP8266, Arduino, ESP32
//#define ZgatewayIR          //ESP8266, Arduino,         Sonoff RF Bridge
//#define ZgatewayBT          //ESP8266, Arduino, ESP32
//#define ZgatewayRF2       //ESP8266, Arduino, ESP32
#define ZgatewaySRFB      //                          Sonoff RF Bridge
//#define ZactuatorONOFF    //ESP8266, Arduino, ESP32,  Sonoff RF Bridge
//#define ZsensorINA226     //ESP8266, Arduino, ESP32
//#define ZsensorHCSR501    //ESP8266, Arduino, ESP32,  Sonoff RF Bridge
//#define ZsensorADC        //ESP8266, Arduino, ESP32
//#define ZsensorBH1750     //ESP8266, Arduino, ESP32
//#define ZsensorTSL2561    //ESP8266, Arduino, ESP32
//#define ZsensorBME280     //ESP8266, Arduino, ESP32
//#define ZsensorDHT        //ESP8266, Arduino, ESP32,  Sonoff RF Bridge
//#define ZgatewayRFM69     //ESP8266, Arduino, ESP32
```
Or did you change the publisch topic yourself in config_SRFB.h?
Default it schould publish on `/SRFBtoMQTT`.
Of course I do get the `LWT Online` and `version 0.7` messages (I am still on v0.7), but I do not get anything on `SYStoMQTT`

From the looks of it you activated the RF IR and BT modules, but you schould only use the SRFB module.

gabr777, 2018-06-28:
My bad. I downloaded the .8 version where I didn't change the defines.
Thanks for pointing that out!
Now it's working.
This is exactly what I wanted!
Thanks!

starkillerOG, 2018-06-28:
@gabr777 Glad it is working for you now.

I guess this issue can now be closed @1technophile ?

starkillerOG, 2018-06-28:
@gabr777 Just curious, does the blue wifi LED do something on you Sonoff RF Bridge R2 after flashing with OpenMQTTGateway?
I am trying to get that LED to show me the wifi connection status.
I think you have to define led_error as GPIO 13 instead of 44, but I have not tryed it yet.
```
/*-------------DEFINE PINs FOR STATUS LEDs----------------*/
#define led_receive 40
#define led_send 42
#define led_error 44
```

1technophile, 2018-06-28:
Yes it can be closed as you validated the compatibility with sonoff rf bridge r2. @gabr777 could you close it please

gabr777, 2018-06-28:
The blue led doesn't do anything now.
I'll change the def and see. Will get back here in a week.

1technophile, 2018-07-28:
@gabr777 if you have some results to share please share it into #221

closed 2018-07-28 by 1technophile