1technophile/OpenMQTTGateway #205: Feature Request: Zigbee Support

jwillmer, 2018-05-18:
Zigbee is becoming the de facto standard in home automation. It would be great if `OpenMQTTGateway` could support this protocol. Below I have a write up about the most promising solution at the moment. I think with some work it would be possible to write an interface for the serial communication of the CC253X hardware. 

There is a [`zigbee2mqtt`](https://github.com/Koenkk/zigbee2mqtt) bridge for Raspberry Pi already. It is using [`zigbee-shepherd`](https://github.com/zigbeer/zigbee-shepherd) that runs on node.js. It all is build around the [CC253X USB stick](http://www.ti.com/tool/cc2531emk). The [cc-znp](https://github.com/zigbeer/cc-znp) library offers a firmware for [serial communication with the Zigbee chip](https://github.com/zigbeer/cc-znp/wiki). A good explanation about the hardware setup can be found [zigbee2mqtt wiki](https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started).

hihp, 2019-01-24:
Any plans on this? I am currently looking into extending my little home automation setup at home, and I would really love to have a ESP8266- or Arduino-based MQTT bridge linking all non-Wifi devices to a Home Assistant instance, so Zigbee support in MQTT would be perfect.

**I actually would be willing to pay money for this, as in setting a substantial bounty for this to be implemented. if necessary via a platform like Gitcoin or so.**

I think that indeed, a TI chip would be easiest - something like the CC2530. There are pretty cheap to get even in a module that includes an actual antenna (search via Google for "CC2530 module"), and using the Z-Stack by TI seem to be farely straightforward to use:

https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/218226

One can even flash a CC2530 without the CCDebugger, using an Arduino or even an ESP8266:

https://koenkk.github.io/zigbee2mqtt/information/alternative_flashing_methods.html

1technophile, 2019-02-24:
Surely zigbee will be an ultimate feature for OMG. Thanks for proposing support it is appreciated. At this time I don't know yet how I will go into this.
For the next release 0.9.1 I will concentrate on Lora and autodiscovery features. Once this is stable zigbee will be an excellent candidate for the next steps.

snovotill, 2019-03-15:
I see the Nordic nRF52 chips such as nRF52840 support **ZigBee**, **BLE5** and **Thread**. The new TI chips CC2652R, CC1352R, CC1352P also. The Nordic also have built-in USB, while the TI support sub-GHz radio.

podarok, 2019-08-05:
I guess instead of copying what zigbee2mgtt does would be cool to merge effort on building gateway framework that merge zigbee2mgtt and OMG all together

james-fry, 2019-10-18:
This is a really interesting idea.
I found this thread since I recently purchased a [TI cc1352p](http://www.ti.com/lit/ds/symlink/cc1352p.pdf) based board for zigbee2mqtt but I am really interested by the sub-1GHz (inc 433 and 868 mhz) of this chipset and wondered on the possibility of OMG type capability in parallel with Z2M. It has the possibility to be an absolutely killer home automation hub.

1technophile, 2019-10-20:
> I recently purchased a TI cc1352p based board for zigbee2mqtt but I am really interested by the sub-1GHz (inc 433 and 868 mhz)

It seems an interesting board, Could you share a link to the board you bought?

james-fry, 2019-10-20:
Sure. I bought this board:
http://www.ti.com/tool/LAUNCHXL-CC1352P

It’s about $50 + tax

There are two versions of the board with different RF tuning for either 2.4ghz or sub 1ghz. I purchased the P-2 version which is more suited for 2.4 for zigbee. That’s said both boards will operate with both bands.

Support for this chipset was only added for zigbee2mqtt in the past week. There is an active github issue thread if you are interested:
https://github.com/Koenkk/zigbee2mqtt/issues/1429

Gamester17, 2020-02-02:
FYI, Tasmota firmware for ESP8266 recently added bridge support for Zigbee via TI CC2530 chip that is flashed with a custom Texas Instruments Z-Stack firmware from Zigbee2mqtt project by Koenkk

https://tasmota.github.io/docs/#/Zigbee

Z2T (Zigbee to Tasmota) concept allows you to make your own Zigbee to WiFi bridge

https://www.cnx-software.com/2019/12/23/tasmota-open-source-firmware-now-supports-zigbee-connectivity/

"Basically all you need is ESP82xx hardware connected to a Texas Instruments C2530 based Zigbee device over a serial. Note that C2531 based device won’t work since they are connected over USB. Normally, C2530 firmware is flashed through CC Debugger tool, but in this case, you don’t need to purchase extra hardware since Z-Stack firmware flashing can be performed through any ESP8266 board running CCLib instead. Once it’s done flash Zigbee2Tasmota Tasmota firmware to your ESP82xx board or device, and you need to configure the system for communication between the Zigbee module and ESP8266. This is all explained in detail in the documentation."

wimpie007, 2020-02-07:
integrating Z2T in home assistant is difficult and involves lots of custom translating stuff for the moment. PITA.

Hedda, 2020-04-20:
FYI, Itead has just launched Sonoff ZBBridge as an inexpensive Sonoff Zigbee Bridge (which is somewhat based on a similar concept/idea as the product that they have in their Sonoff RF Bridge)

- https://www.cnx-software.com/2020/04/16/sonoff-zbbridge-wifi-to-zigbee-gateway/

This new Sonoff ZBBridge (and their existing Sonoff RF Bridge) uses ESP8266 ("ESP8266EX") for WiFi and bridge/gateway/controller software, so the main difference is that they use different types of MCU for the RF radio module part.

- https://www.itead.cc/sonoff-zbbridge.html

According to the teardown on notenoughtech.com it sounds as if it the new Sonoff ZBBridge is based on Silicon Labs EFR32 (Mighty Gecko) for Zigbee 3.0 radio module support 

- https://notenoughtech.com/home-automation/sonoff/sonoff-zigbee-bridge-preview/

As you probably already know, Tasmota does something similar with Zigbee2Tasmota but by connecting an ESP8266 to a Texas Instrument CC2530 module instead however it too is using a serial communication protocol, see:

- https://tasmota.github.io/docs/Zigbee/

Hedda, 2020-04-20:
One workaround for Home Assistant users could be to simply present a Zigbee radio module as a remote adapter to Home Assistant's ZHA (their Zigbee Home Automation integration component)?

- https://www.home-assistant.io/integrations/zha/

You could use something like ser2net to allow ZHA to remote connect to it via serial over network, like:

- https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

Then you need a radio library for zigpy that works with the serial protocol used. For example with that Sonoff ZBBridge you would need a zigpy radio library for EZSP protocol used by Silicon Labs EFR32 

- https://github.com/zigpy/zigpy

Hopefully, their bellows radio library should already be compatible or could be made to be if updated

- https://github.com/zigpy/bellows

zigpy devs already do something similar to connect remotely with ESP8266 based "ZiGate Pack WiFi"

- https://zigate.fr/produit/zigate-pack-wifi-v1-3/
- https://github.com/fairecasoimeme/ZiGate/
- https://github.com/zigpy/zigpy-zigate

wimpie007, 2020-04-20:
"As proof-of-concept you could do this with a CC2530 remote to zigpy-cc from a ESP8266 using ser2net"
I use a esp8266 with CC2530 altready to connect to a zigbee2mqtt instance.
Espeasy with serial server. 
Works like a threat.

wimpie007, 2020-04-20:
the new chipset seems promising!! (range!)

Hedda, 2020-04-20:
Do you think that you could maybe use the serial bridge from esp-link for a serial forwarding feature?

- https://github.com/jeelabs/esp-link
  - https://github.com/jeelabs/esp-link/tree/master/serial

Hedda, 2020-04-20:
oxan might have a "stream server" solution for ESPHome firmware in https://github.com/esphome/feature-requests/issues/660

1technophile, 2020-04-20:
The best way could be to share a common zigbee driver on all these projects, I have some zigbee devices around and a CC2530, this will be the occasion to play in the next months

Hedda, 2020-04-20:
@1technophile problem with that is that there are so many different Zigbee modules 

Maybe better to port zigpy and its radio libraries or would that be overkill?

- https://github.com/zigpy/zigpy

- https://github.com/zigpy/

1technophile, 2020-04-20:
If we cover the CC2530 on an ESP you don't think it would be enough?

james-fry, 2020-04-20:
The zigbee2mqtt project is beginning to move away from the cc2530. The cc2530 is a low power (CPU) device and has limitations as a zigbee coordinator.

Hedda, 2020-04-20:
As james-fry wrote; Zigbee2mqtt is long moving towards more powerful Zigbee 3.0 compatible radio modules with support for CC26X2R1 (CC2652/CC2652R/CC2652P) and CC1352P-2, but more importantly Zigbee2mqtt is also moving towards hardware-independent support for ConBee, ConBee II, and RaspBee support in Zigbee2mqtt already available. 

It would also not work on Sonoff ZBBridge as I requested because it contains a Silicon Labs EFR32MG21 Zigbee 3.0 radio module support. EFR32MG21 is one of the latest chip in EFR32 Gecko Series 2 MCU

Zigpy has radio libraires to support for all of these mentioned radio modules and more.

Hedda, 2020-04-20:
Checkout https://github.com/Koenkk/zigbee2mqtt/issues/1429

Guys there recommend different CC2652 based modules if you have to choose one. The downside is that a single such CC2652 baded module will still cost you more than a Sonoff ZBBridge.

Again, Sonoff ZBBridge is a prepackaged product at less than $17 so hard to beat that price

james-fry, 2020-04-20:
The CC1352P-2 is I think especially interesting for omg as it can also do sub 1ghz as well a 2.4ghz (zigbee)

Hedda, 2020-04-21:
I also believe that those sub 1GHz modules do not do two or more frequencies at the same time(?) as I believe that you set it to one specific frequency (like 2.4GHz, 915MHz or 868MHz) to use it(?)

Are sub 1GHz even used by any Zigbee application profiles part of the Zigbee 3.0 spec in any case? 

https://www.unifore.net/home-alarm-system/introduction-to-zigbee-3-0-zigbee-ha-protocol.html

- Zigbee Home Automation (Zigbee HA)
- Zigbee Light Link (Zigbee LL)
- Zigbee Building Automation (Zigbee BA)
- Zigbee Retail Services (ZigBee RS)
- Zigbee Health Care (Zigbee HC)
- Zigbee Telecommunication services (Zigbee TS)

Or is sub 1GHz only used by Zigbee application profiles that are not part of the Zigbee 3.0 spec?

- Zigbee Smart Energy (Zigbee SE)

Hedda, 2020-04-21:
Anyway, the CC2652P looks like the best chip based on chip specification, with CC2652R and CC2652RB chips having the second-best chip specifications, however, CC1352P based modules are much more readily available today.

Both CC1352P and CC2652P/CC2652R/CC2652RB chips can be bought directly to TI but there appear not to be any complete CC2652 based modules available yet, while there already are already CC1352P modules available from example Ebyte.

http://www.ebyte.com/en/product-view-news.aspx?id=766

Search for article: "E79-400DM2005S" on sites like eBay or Aliexpress to find Ebyte CC1352P module.

Again, CC1352 and CC2652 just users a newer version of the same serial protocol and command set as the CC253x series of chips, so if you get it working on either C1352 or CC2652 then it should not be too much work updating the solution to work on all. 

See zigpy-znp https://github.com/zha-ng/zigpy-znp and zigpy-cc https://github.com/zigpy/zigpy-cc

Hedda, 2020-04-21:
To get the equivalent module from Silicon Labs then might want to look at the EFR32 Series 1 like the EFR32MG1B SoC.

Ebyte's E180-ZG120A and E180-ZG120B are Zigbee 3.0 module based on EFR32MG1B SoC

http://www.ebyte.com/en/product-view-news.aspx?id=842

Both E180-ZG120A and E180-ZG120B modules can be found on the same sites for a similar price.

I also found this inexpensive development board based on an E180-ZG120B module from Ebyte:

https://www.aliexpress.com/item/4000538003718.html

Looks like you can just remove some jumpers to disable the USB converter to use the serial directly.

See bellows for zigpy's Silicon Labs library https://github.com/zigpy/bellows

Hedda, 2020-04-23:
Almost any ESP8266 development board (like a NodeMCU or a Wemos D1 Mini) with a Silicon Labs EFR32MG based Zigbee module like the Ebyte E180-ZG120B would just about be about the equivalent to having a hacked Sonoff ZBBridge, so such a setup could be used as a development environment.

The downside to using such setup instead of Sonoff ZBBridge is that the E180-ZG120B module will likely not come preloaded with firmware so have to build EmberZNet PRO Zigbee Stack firmware with coordinator device type config and flash firmware to the E180-ZG120B module first.

See a longer discussion about building EmberZNet firmware on https://github.com/xoseperez/espurna/issues/2224

Hedda, 2020-04-28:
FYI @SillyDay don't own a E180-ZG120B but has tried to build a firmware for it as per discussion here:

https://github.com/Koenkk/zigbee-herdsman/issues/168

SillyDay GitHub repo for EFR32 firmware:

https://github.com/SillyDay/EFR32

Hedda, 2020-05-05:
FYI, as per related request https://github.com/zigpy/zigpy/issues/405 I just learned that there is an existing ESP8266-based networked-attached Zigbee-adapter called "ZiGate Pack WiFi adapter" which has a new v2.0 firmware that archives this requested function of UART-to-TCP/IP (for Serial-port to WiFi-bridge function) using "ESP-LINK from Jeelab" software, in addition, using ESP-LINK also adds mDNS zeroconf to allow automatic network discovery and configuration:

- https://github.com/jeelabs/esp-link

Description of functions that using ESP-LINK will add to v2.0 firmware for ZiGate Pack WiFi adapter:

https://translate.google.com/translate?sl=fr&tl=en&u=https%3A%2F%2Fzigate.fr%2Fdocumentation%2Fdescription-du-firmware-v2-xx%2F

- ZiGate TCP 9999 to UART WiFi Bridge
- WiFi configuration in client and access point mode
- DHCP client or static IP
- Client syslog
- SNTP client
- mDNS
- MQTT client
- WiFi firmware update (OTA)
- Console for logging and validating the operation of the ZiGate

As I understand it, all ZiGate hardware look to be modular in design and the "ZiGate Pack WiFi adapter" is really just an optional ESP8266 based "dumb" UART-to-TCP/IP (for Serial-port to WiFi-bridge function) for the standard "ZiGate TTL adapter" that allows users to connect to it remotely using TCP/IP over your home LAN (Local Area Network) instead of plugging it directly to your computer via the optional USB adapter. 

- https://zigate.fr/produit/module-wifi-v1-3-compatible-zigate/

Specifically, please see the picture of "ZiGate Pack WiFi adapter" https://zigate.fr/produit/zigate-pack-wifi-v1-3/ compared to the picture of "ZiGate TTL USB adapter" https://zigate.fr/produit/zigate-ttl/

Thus "ZiGate Pack WiFi adapter" allows ZHA users to have a networked Zigbee adapter setup like this:

ZHA <–> zigpy/zigpy-zigate <–> TCP/IP over LAN <–> ZiGate-WiFi <–> UART <–> ZiGate Radio

Suggesting this now as I just learned from @doudz there that the new v2.0 version of the ZiGate Pack WiFi adapter firmware contains "ESP-LINK from Jeelab" software which among other things adds mDNS and  UART WiFi Bridge support over TCP. As I understand, version v1.x of the firmware for the ZiGate Pack WiFi adapter basically only contained a simple UART/serial-port server forwarding service ( serial server software that just acts as a dumb Zigbee to WiFi bridge for zigpy-zigate), while the new version v2.x also has more advanced features (which does not need to used) it still also contain a simple UART/serial-port server forwarding service, but now mDNS also makes it easier to discover the adapter on your local network.

Now it would be awesome if the ZHA integration component for Home Assistant from an end-user perspective supported just as an easy detection and configuration of network networked-attached Zigbee coordinator adapters, like the Sonoff ZBBridge and the ZiGate Pack WiFi adapter. 

I would therefore also suggest using some kind of [Zero-configuration networking (zeroconf)](https://en.wikipedia.org/wiki/Zero-configuration_networking) method, like for example [mDNS](https://en.wikipedia.org/wiki/Multicast_DNS), (as mDNS is already in use in Home Assistant Core), to make the ZHA integration component for Home Assistant automatically detect, connect, and configure compatible networked-attached Zigbee coordinator adapters like the "ZiGate Pack WiFi adapter" as that is otherwise already supported by the zigpy-zigate radio library for zigpy.

Hedda, 2020-06-01:
FYI, @s-hadinger has now got a Sonoff ZBBridge and started reviewing the board in it (for Tasmota)

- https://github.com/arendst/Tasmota/issues/8583

First signs it is the EFR32 Zigbee module it has inside it has EmberZNet / Ember based firmware.

1technophile, 2020-06-01:
Thanks for the info!

Hedda, 2020-06-02:
There is now also a deep dive follow-up discussion here in parallel about hacking or flashing its EFR32:

- https://github.com/Koenkk/zigbee-herdsman/issues/168

Hedda, 2020-06-22:
FYI; @s-hadinger is making great progress hacking the Zigbee module in Sonoff ZBBridge for Tasmota:

- https://github.com/arendst/Tasmota/issues/8583
  - https://github.com/arendst/Tasmota/pull/8709
  - https://github.com/arendst/Tasmota/pull/8739

@mtx512 has now built a custom EZSP NCP firmware for the EFR32 module inside Sonoff ZBBridge:

- https://github.com/mtx512/efr32
  - https://github.com/mtx512/efr32/tree/master/SonoffZigbeeBridge

Interestingly @s-hadinger also added a TCP serial bridge to Tasmota firmware for remote access:

- https://github.com/arendst/Tasmota/pull/8702

> ## Description:
> Add a new feature: Serial to TCP bridge (similar to esp-link). This allows to remotely communicate to a MCU through ESP8266. Needs `#define USE_TCP_BRIDGE`
> 
> It adds 2 GPIO types: `TCP Tx (208)` and `TCP Rx (209)` and can work with hardware or software serial.
> 
> Commands:
> 
> * `TCPBaudRate <x>`: sets the baudrate for serial (only 8N1 mode), min `1200`, max `115200` by 1200 increments.
> * `TCPStart <port>`: listens to port `<port>`. This features supports 2 parallel TCP connexions, which can be useful if you need a terminal + a specific protocol (like XMODEM). The 3rd connection will disconnect an previous connection. The number of parallel connections is a compile-time option.
> * `TCPStart 0` or `TCPStart`: shuts down the TCP server and disconnects any existing connection.
> 
> For security reasons, the TCP bridge is not started at boot, and requires an explicit `TCPStart` command (can be automated with Rules).

Hedda, 2020-06-23:
FYI, Silicon Labs recently just released a less expensive "EFR32xG22 Wireless Gecko Starter Kit" (SLWSTK6021A) development kit for $99 which only contain one mainboard/development board and a couple of EFR32MG22 modules which are low-power Zigbee Green Power compatible Mightly Gecko Series 2 radios.

https://www.silabs.com/products/development-tools/wireless/efr32xg22-wireless-starter-kit

While the EFR32MG22 low-power radios it comes with are not really usable as Zigbee coordinators, for $99 that kit can be very useful to developers since it jas been confirmed that buying that kit also gives full official access to Silicon Labs Zigbee Stack, Zigbee SDK, and libraries (current, future, and old versions).

Hedda, 2020-06-24:
> FYI, Silicon Labs recently just released a less expensive "EFR32xG22 Wireless Gecko Starter Kit" (SLWSTK6021A) development kit for $99 which only contain one mainboard/development board and a couple of EFR32MG22 modules which are low-power Zigbee Green Power compatible Mightly Gecko Series 2 radios.
> 
> https://www.silabs.com/products/development-tools/wireless/efr32xg22-wireless-starter-kit
> 
> While the EFR32MG22 low-power radios it comes with are not really usable as Zigbee coordinators, for $99 that kit can be very useful to developers since it has been confirmed that buying that kit also gives full official access to Silicon Labs Zigbee Stack, Zigbee SDK, and libraries (current, future, and old versions).

Those who are interested in that official $99 SLWSTK6021A started kit might also be interested in buying article SLWRB4180A for around $49 at the same time as that is the matching EFR32MG21 based high-power (+20 dBm) radio board for that dev board kit.

https://www.silabs.com/products/development-tools/wireless/slwrb4180a-efr32-wireless-gecko-radio-board

(Google search shows that SLWRB4180A can be bought from many of resellers globally who also stock the SLWSTK6021A kit).

Hedda, 2020-07-03:
FYI, @s-hadinger has after the last pull request posted saying that he now feels Tasmota ready for testing of EZSP v8 interface:

https://github.com/arendst/Tasmota/issues/8583

https://github.com/arendst/Tasmota/pull/8839

Not for everyone yet since still need to flash its EFR32 chip with some type of J-Link Debug Probe (might always be required?).

Hedda, 2020-09-11:
Any further thoughts on Sonoff ZBBridge support now that this is fully supported and already very stable feature in Tasmota?

https://github.com/arendst/Tasmota/issues/8583

That Sonoff ZBBridge is inexpensive but contain great Zigbee hardware with potential for different types of Zigbee controller uses, as inside it has an ESP8266 WiFi SoC/MCU connected via UART to a very powerful Silicon Labs EFR32MG2 (Mighty Gecko Series 2) Zigbee SoC/MCU on the same board.

https://sonoff.tech/product/smart-home-security/zbbridge

Tasmota allows using its ser2net like TCP serial-server inside ESP8266 firmware to take direct control of the Zigbee chip that it is connected to via UART/serial and then connect to Zigbee devices without any further involvement on Tasmota (which via its TCP serial-server only will only act as a dumb pass-through WiFi-to-serial bridge without any knowledge of what the data traffic that is going back and forth between the remote Zigbee chip AND different home automation software like example Home Assistant).

You can even make your own DIY WiFi bridge with an ESP8266 board + an ICC-A-1 module torn out of an IKEA Tradfri device:

https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module

Example with ZHA integration component in Home Assistant which add native support for Zigbee adapters, including the ZBBridge: 

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

Sure, it could have been even greater with ESP32 but still geat ESP8266 hardware with Silabs EFR32 MG2 Zigbee coordinator!

1technophile, 2020-09-11:
It would be an interesting addition to OMG, but maybe the best way would be to externalize it into a Zigbee library. This way it will benefit the different projects (OMG, ESPurna, ESPEasy, ESPHome...).

Hedda, 2020-09-11:
> It would be an interesting addition to OMG, but maybe the best way would be to externalize it into a Zigbee library. This way it will benefit the different projects (OMG, ESPurna, ESPEasy, ESPHome...).

zigpy and its radio libraries are externalized for use over serial-port via any pass-through serial-bridge if that is what you mean? 

https://github.com/zigpy/zigpy/

Or do you mean splitting all the Zigbee code parts out on Zigbee2Tasmota (Z2T) into an Arduino library which could be reused? 

https://tasmota.github.io/docs/Zigbee/

As I understand it, the Zigbee2Tasmota (Z2T) solution is very large, especially compared to to only providing a TCP serial-server.

1technophile, 2020-09-16:
> Or do you mean splitting all the Zigbee code parts out on Zigbee2Tasmota (Z2T) into an Arduino library which could be reused?

Exactly. This way it could be used by other projects.

Hedda, 2020-09-16:
> > Or do you mean splitting all the Zigbee code parts out on Zigbee2Tasmota (Z2T) into an Arduino library which could be reused?
> 
> Exactly. This way it could be used by other projects.

What I know is that whole Zigbee bridge code in Zigbee2Tasmota (Z2T) is large and very resource-intensive for a ESP8266 MCU, and that why they now provide a separate dedicated tasmota-zbbridge firmware for that where they disable other stuff not needed for Zigbee bridge. ESP32 would be better for such Zigbee bridge functions: https://tasmota.github.io/docs/Zigbee/

The whole Zigbee bridge code is not needed though to only provide a stable TCP serial-server (similar to ser2net) to enable transparent pass-through function over IP for https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

clubbit, 2020-10-05:
https://slsys.github.io/Gateway/README_rus.html

jwillmer, 2020-10-05:
Cool project @clubbit but what does it have to do with this task? Feels like attention spam 👎

1technophile, 2020-10-05:
Thanks @jwillmer, link removed

Hedda, 2021-03-11:
FYI, HW tip now is to instead use Tube's Zigbee Gateways open-source hardware by @tube0013 (which is tested with ESPHome).

It is based on WT32-ETH01 ESP32 board from Wireless-Tag which also has wired Ethernet.

He designed two variants; one has a Silicon Labs EFR32 Series 2 module and one that has a Texas Instruments CC2652P module.

https://github.com/tube0013/tube_gateways

[https://github.com/zigpy/zigpy/discussions/584](https://github.com/zigpy/zigpy/discussions/584)

https://www.tubeszb.com/shop/coordinators/2

[image]

[image]

Hedda, 2021-03-19:
I guess that best would probably be if you could get some informal advice about their API/CLI protocols from @s-hadinger as he has experience of both since he implemented ZNP (TI Z-Stack) and EZSP (Silabs EmberZNet) support in Tasmota's Zigbee2Tasmota (Z2T):

https://tasmota.github.io/docs/Zigbee/

If only look at availability of dev-tools + modules available to buy then Texas Instruments CC2652P and/or CC1352P are best today:

https://www.szrfstar.com/product/list-196-en.html

https://www.ebyte.com/en/product-view-news.html?id=1002

Silabs require buying their official starter kit to get access to firmware SDK + availability of EFR32MG21 modules is currently low:

https://www.silabs.com/development-tools/wireless/efr32xg22-wireless-starter-kit

https://www.silabs.com/development-tools/wireless/slwrb4180b-efr32xg21-wireless-gecko-radio-board

https://github.com/zigpy/zigpy/discussions/586

The older EFR32MG12 is however available and it is even possible to extract such modules from any IKEA Trådfri device:

https://www.ebyte.com/en/product-view-news.html?id=842

https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module

Texas Instruments currently also have more open source tools for things like firmware flashing + NVRAM backup and restore:

https://github.com/JelmerT/cc2538-bsl

https://github.com/zigpy/open-coordinator-backup

https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md#backup-and-restore

https://github.com/zigpy/zigpy-znp

PS: I believe @tube0013 and @grobasoz (who both make DIY gateway hardware) mention they prefer TI CC chips, tools, FW, APIs?

Hedda, 2021-03-19:
CC1352 is more expensive but allow Sub-1 GHz frequencies used by Zigbee Smart Energy (ZSE) profile which is not part of Zigbee 3.0

https://www.szrfstar.com/product/235-en.html

https://www.szrfstar.com/product/188-en.html

Zigbee Smart Energy is the world standard used by electricity, water, and gas meters that all utility companies install in your house.

https://zigbeealliance.org/solution/smart-energy/

https://zigbeealliance.org/wp-content/uploads/2019/11/docs-07-5356-19-0zse-zigbee-smart-energy-profile-specification.pdf

https://training.ti.com/overview-zigbee-smart-energy-internet-things

https://www.ti.com/lit/slaa467

https://www.ti.com/lit/pdf/tidu213

https://www.ti.com/lit/pdf/swra467

https://www.nxp.com/pages/jn516x-zigbee-smart-energy:ZIGBEE-SMART-ENERGY

"ZigBee Smart Energy (SE) is the world's leading standard for interoperable wireless products that monitor, control and automate the delivery and use of energy (and other resources, such as water). A wireless network is implemented at the consumer's premises using the ZigBee PRO protocol with the Smart Energy application profile. This Home Area Network (HAN) contains devices such as a Metering Device, In-Premise Display (IPD) and Load Control Device, and is connected to the energy provider using an Energy Service Interface (ESI) via a backhaul network."

Hedda, 2021-03-26:
> > > It would be an interesting addition to OMG, but maybe the best way would be to externalize it into a Zigbee library. This way it will benefit the different projects (OMG, ESPurna, ESPEasy, ESPHome...).
> > > 
> > Or do you mean splitting all the Zigbee code parts out on Zigbee2Tasmota (Z2T) into an Arduino library which could be reused?
> > 
> Exactly. This way it could be used by other projects.

@1technophile might interested in today's related discussion between @Barracuda09 and @TD-er c in https://github.com/letscontrolit/ESPEasy/issues/3021

Hedda, 2021-09-06:
FYI, thegroove made a simple to use Serial-to-IP bridge appliance firmware for Sonoff Zigbee Bridge running ESPHome firmware:

https://github.com/thegroove/esphome-zbbridge

oxan and thegroove also look to have made two alternative serial server code component solutions for ESPHome firmware:

https://github.com/oxan/esphome-stream-server

https://github.com/thegroove/esphome-serial-server

thegroove  even made an experimental version with Zeroconf support for automatic network discovery by ZHA in Home Assistant:

https://github.com/thegroove/esphome-zha-ezsp-zeroconf

https://www.home-assistant.io/integrations/zha/#discovery-via-usb-or-zeroconf

Note that Zigbee2MQTT is another popular alternative to Home Assistant's ZHA for use with this type of Serial-to-IP bridge:

https://www.digiblur.com/2021/03/zigbee2mqtt-with-sonoff-zigbee-bridge.html

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

Hedda, 2021-09-22:
FYI, ZB-GW03 eWeLink Ethernet Zigbee Gateway is based on an ESP32 and Silabs EFR32 Zigbee and has been hacked with Tasmota.

https://github.com/arendst/Tasmota/discussions/12764

ZB-GW03 has been hacked with Tasmota32 and look to support the same Tasmota and Silabs firmware as ITead Sonoff ZBBridge, however since the ZB-GW03 has wired Ethernet port instead of WiFi it should be a better option compared to ITead Sonoff ZBBridge.

https://templates.blakadder.com/ewelink_ZB-GW03

[image]

[image]

[image]

ZB-GW03 v1.0 and ZB-GW03-V1.2 is apparently sold rebranded under many names, including SmartWise and EACHEN brands:

https://www.okosabbotthon.hu/smartwise-zigbee-bridge-pro

https://www.amazon.de/APP-Fernbedienung-Sprachsteuerung-Funktioniert-Verbindung-Smart-Produkten/dp/B094JKVLNR/

[image]

[image]

Looks like is it is based on the same "SM-011 V1.0" module by CoolKit as ITead uses in their Sonoff ZBBridge Zigbee Bridge:

https://github.com/zigpy/zigpy/discussions/586

https://www.coolkit.cn/product/sm-011/

https://github.com/CoolKit-Technologies/DevDocs/tree/master/Zigbee

https://github.com/CoolKit-Technologies/DevDocs/blob/master/Zigbee/SM-011%E5%BA%94%E7%94%A8%E6%8C%87%E5%AF%BC%E4%B9%A6.md

https://translate.google.com/translate?hl=&sl=auto&tl=en&u=https%3A%2F%2Fgithub.com%2FCoolKit-Technologies%2FDevDocs%2Fblob%2Fmaster%2FZigbee%2FSM-011%E5%BA%94%E7%94%A8%E6%8C%87%E5%AF%BC%E4%B9%A6.md

Hedda, 2021-11-01:
FYI, there's now a detailed step-by-step instruction about flashing Tasmota on ZB-GW03 eWeLink Ethernet Zigbee Gateway (wired network bridge) for Zigbee2Tasmota (also using with Home Assistant's ZHA):

https://thehelpfulidiot.com/a-wired-sonoff-zigbee-alternative

Again, basic info for experienced Tasmota and Z2T users here :

https://templates.blakadder.com/ewelink_ZB-GW03

and again there is also a discussion about it in Home Assistant community forum here:

https://community.home-assistant.io/t/zb-gw03-ewelink-ethernet-zigbee-gateway-now-hacked-with-tasmota-zbbridge-so-can-be-used-via-mqtt-or-as-a-remote-zigbee-adapter-with-home-assistant-zha/341223

Hedda, 2022-05-16:
FYI, ESP32 and CC2652P based "Itead SONOFF Zigbee Bridge Pro" (ZBBridge-P, a.k.a. "ZB Bridge-P") is now available for purchase:

https://itead.cc/product/sonoff-zigbee-bridge-pro/

FCC leak from earlier in January this year showed inside with ESP32 + CC2652P based Sonoff ZBBridge-P v1.1 Zigbee Bridge:

[https://fccid.io/2APN5ZBBRIDGEP](https://fccid.io/2APN5ZBBRIDGEP)

Both NotEnoughTech and digiblur have received theirs and while neither of them has tried to flash/hack it yet they posted some early thoughts on the original eWeLink hub firmware and hardware board, confirming that the first retail version ITead's SONOFF Zigbee Bridge Pro with PCB board labelled "ZBBridge-P V1.1 2021.11.12" is indeed based on Espressiff  ESP32-DOWD-V3 (that ESP32 variant by the way also has integrated support for Bluetooth 4.2 including BLE), Texas Instruments CC2652P Zigbee module labelled "SM-031 V1.1 2021.07.28" (so probably manufactured by [CoolKit Technologies](https://github.com/CoolKit-Technologies/DevDocs/tree/master/Zigbee) like the Zigbee module for previous Sonoff bridge), with a cell-battery for the RTC (Real-Time-Clock), a built-in buzzer for chime or alarm-siren, and also an Espressiff branded PSRAM (ESP-PSRAM16H) with 2MB/16Mbit capacity.

digiblurDIY also mention that he managed to backup original firmware from ESP32 without issue so suspect that it is not locked:

https://www.youtube.com/watch?v=P5op57y5i6k?t=920&ab_channel=digiblurDIY

https://notenoughtech.com/home-automation/sonoff-zigbee-bridge-pro/

https://www.youtube.com/watch?v=fsIjFxCPJLA&ab_channel=NotEnoughTECH

Internal pictures posted from FCC show using an unannounced Texas Intruments CC2652P (CC2652P1) MCU chip based “SM-031 v1.1” radio module which must be made by [CoolKit](https://github.com/CoolKit-Technologies/DevDocs/tree/master/Zigbee) (the company that makes their EFR32MG21 based “[SM-011 V1.0](https://www.coolkit.cn/product/sm-011/)” module).  

The new module based on Texas Instruments CC2652 probably come preloaded with Zigbee Coordinator firmware from Koenkk:

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0

Again the main downside the this new and the old Sonoff ZBBridge is that they only have WiFi and not a wired Ethernet LAN port.

michcfr, 2024-06-16:
Hello,
A little up for this topic.

Finally, is it expected an OpenMQTTGateway for ZigBee either based on the TI ZNP stack (as done by Tasmota) or based on the Espressif’s Zigbee SDK (built on top of the Zboss stack) for the ESP32-C6?

Regards