1technophile/OpenMQTTGateway #208: Feature Request: Energymonitoring

betonishard, 2018-05-26:
Hi all,
Great project (again). I was wondering If there is a need for adding an LM393 tot OpenMQTT.. I would use it like in this project: https://www.mysensors.org/build/pulse_power

I have a gateway set up in my electrical room. Would it be possible to add this, or manually replace one.?
Regards,

betonishard, 2018-06-03:
Maybe to make it more generic. A pulse counter

1technophile, 2018-07-28:
Yes if someone has this kind of sensor don't hesitate [to develop ](https://github.com/1technophile/OpenMQTTGateway/blob/master/CONTRIBUTING.md)a new ZsensorLM393 and a config_LM393 so as to intégrate it into OMG.

betonishard, 2018-08-02:
That would be great, if only there was someone,...