1technophile/OpenMQTTGateway #210: Feature Request:BLE scan custom devices?

huexpub, 2018-05-31:
great A  project， thanks, i do it work for 2 xiaomi bluetooth temp&humi,but BT scan my Bluetooth  keybroad ,Bluetooth  IR controler，Bluetooth headset，so the temp and humi update too long , if only  scan  the ble mac address of manual set，The efficiency of scan will be higher ?

tunip, 2018-07-24:
Specifying the BLE mac addresses for scanning make sense. If you have multiple ESP32 in the house to cover all BLE devivces you scan sometimes the device multiple times.

thundergreen, 2018-07-27:
What is the best setting for ibeacon presence detection? Bt scan interval to 1 second? It takes ages to recognise my rssi for my beacon . Wanna speed this up.hiw can I do though ?

codegrau, 2018-09-13:
@1technophile Would that feature make it possible to "scan" for presence detection with iPhones without an external app like https://github.com/andrewjfreyer/monitor this bash script on Raspberry Pi would do?

1technophile, 2019-05-04:
white list and black list of ble devices has been introduced in V0.9.1
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-bluetooth-BLE-09#setting-a-white-or-black-list

closed 2019-05-04 by 1technophile