1technophile/OpenMQTTGateway #212: Problems in using RF2

joe63, 2018-06-06:
Hallo, frist of all, thank you for your good work!

I have installed OpenMQTTGateway on an ESP32, to connect a Remotesensor (a 433MHZ Sensor with temperature and humidity) with Openhab.

[image]

First I tested the 433MHZ-Receiver with a SONOFF® 433MHZ Remote Controller and the 

[image]

When I used ZgatewayRF and pushed a Button I saw the message in MQTTlens. Then I saw sending the Remotesensor, but nothing in MQTTlens.

[image]

Then I changed from ZgatewayRF to ZgatewayRF2(with no Change of the wiring). I saw neither a message from the pushed a Button nor from the Remotesensor.

Now I have 2 questions:
1.) why I don't see the message from the pushed button with ZgatewayRF2.
        _Receiving RF codes with the KAKU protocol is not compatible with ZgatewayRF_ 
2.) why I don't see the message from remotesensor with ZgatewayRF2.

Thank you

OpenMQTTGateway version used (V0.7.)
RCSwitch library V2.6.2)

1technophile, 2018-06-07:
Hello,

ZgatewayRF2 support KAKU protocol it is normal that you don't see the push of your remote control. Your remote control is compatible with ZgatewayRF only.
Here is the [list of compatible devices](https://docs.google.com/spreadsheets/d/1DdtVtSsN25nwP6BZI5q6C9yDGz37tUWjw2SQ1RGwBxU/edit?usp=sharing) with ZgatewayRF

Neither ZgatewayRF and RF2 are compatible with weather stations.

If you want a compatible weather station with OpenMQTTGateway I suggest you to take a look to BLE ones like [Mi flora](https://www.banggood.com/Original-Xiaomi-4-in-1-Flower-Plant-Light-Temperature-Tester-Garden-Soil-Moisture-Nutrient-Monitor-p-1068369.html?rmmds=myorder&cur_warehouse=CN) or [Mi jia](https://www.banggood.com/Xiaomi-Mijia-Bluetooth-Thermometer-Hygrometer-with-LCD-Screen-Magnetic-Suction-Wall-Stickers-p-1232396.html?rmmds=myorder&cur_warehouse=CN)
They are compatible with ESP32 without any added component

joe63, 2018-06-07:
Hello,

thank you for your fast response and for the list of comp devices.  

My 'problem' is that I have serveral remote sensors and I wanted use them......
Which code(s) do weather stations use?

Is it possible to use 5 Mi jia at the same time(not at the same second) on one iobrocker?

1technophile, 2018-06-07:
5 mi jia should work at the same time, never tested but I don't see any reason so that it doesn't work.
To read uour current remote sensors you should take a look to RFlink

1technophile, 2018-06-14:
I'm closing this issue if you have other questions don't hesitate to ask questions on the gitter chat room

closed 2018-06-14 by 1technophile