1technophile/OpenMQTTGateway #213: Compiling error

Salvurion, 2018-06-08:
Hi there,

I want to setup an esp8266 with 433Mhz transmitter and receiver for my lights, and control them with my home assistant. but when i want to compile the code to the esp i get an error. 

"C:\Program Files (x86)\Arduino\portable\sketchbook\OpenMQTTGateway\OpenMQTTGateway.ino: In function 'void setup()':

OpenMQTTGateway:348: error: 'setupRF' was not declared in this scope

     setupRF();

             ^

OpenMQTTGateway:354: error: 'setupRF2' was not declared in this scope

     setupRF2();

              ^

C:\Program Files (x86)\Arduino\portable\sketchbook\OpenMQTTGateway\OpenMQTTGateway.ino: In function 'void loop()':

OpenMQTTGateway:611: error: 'RFtoMQTT' was not declared in this scope

       if(RFtoMQTT()){

                   ^

OpenMQTTGateway:627: error: 'RF2toMQTT' was not declared in this scope

       if(RF2toMQTT()){

                    ^

C:\Program Files (x86)\Arduino\portable\sketchbook\OpenMQTTGateway\OpenMQTTGateway.ino: In function 'void receivingMQTT(char*, char*)':

OpenMQTTGateway:801: error: 'MQTTtoRF' was not declared in this scope

   MQTTtoRF(topicOri, datacallback);

                                  ^

OpenMQTTGateway:804: error: 'MQTTtoRF2' was not declared in this scope

   MQTTtoRF2(topicOri, datacallback);

                                   ^

exit status 1
'setupRF' was not declared in this scope"

I'm not sure what to change, if someone could help me.

1technophile, 2018-06-08:
Did you checked the wiki:
https://github.com/1technophile/OpenMQTTGateway/wiki

and the troubleshooting section?
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

Salvurion, 2018-06-08:
Hello,
the troubleshooting section says that i cant have an Compilation error when i use "the versions of the libraries present into the lib folder" i checked and i think i got all of them. here an screenshot.

[image]

1technophile, 2018-06-08:
could we continue the discussion in the gitter tchat please 
https://gitter.im/OpenMQTTGateway/Questions_support

closed 2018-06-08 by 1technophile