1technophile/OpenMQTTGateway #214: RF + IR = bad IP

dkmatt0, 2018-06-12:
Hello,

(sorry for bad english, I'm french)

If I use only ZgatewayRF, no problem.
But, if I use ZgatewayRF AND ZgatewayIR, I have a problem with IP address (in serial monitor):
`` 
    OpenMQTTGateway ip:
     192.168.10.3
     Ethernet ok
     1883
     Connecting to MQTT by IP adress
     19~.168.10.2
``

See the last line, in config file, it's 19**2**.168.10.2, not 19~.168.10.2
No problem if I use only ZgatewayIR

I work on Arduino Uno R3, also, I have been tested on a compatible arduino pro mini, same problem.
I use the last beta version (commit 14d8eef7cd0dd686ceffe8167a382e11e1ece45a), same problem with V0.7

1technophile, 2018-06-12:
What is the remaining memoiry you get after compiling on arduino uno, I suspect you don't have enough

dkmatt0, 2018-06-13:
Global variables use 1717 bytes (83%) of dynamic memory, leaving 331 bytes for local variables. Maximum is 2048 bytes.

It is not enough ?

dkmatt0, 2018-06-16:
Sketch uses 29164 bytes (94%) of program storage space. Maximum is 30720 bytes.
Global variables use 1529 bytes (74%) of dynamic memory, leaving 519 bytes for local variables. Maximum is 2048 bytes.

Do you want that ?

1technophile, 2018-06-16:
Strange when I compile for arduino uno I have an error message saying there is not enough memory:

> Le croquis utilise 29260 octets (90%) de l'espace de stockage de programmes. Le maximum est de 32256 octets.
> 
> Les variables globales utilisent 2198 octets (107%) de mémoire dynamique, ce qui laisse -150 octets pour les variables locales. Le maximum est de 2048 octets.
> Mémore insuffisante ; consulter la page http://www.arduino.cc/en/Guide/Troubleshooting#size pour obtenir des astuces sur comment le réduire.
> Erreur de compilation pour la carte Arduino/Genuino Uno

It is odd that you are able to compile without error if you have IR and RF activated.

Please note that I don't think it is a good idea to use a UNO for more than one gateway. If you want to use two gateways I advise to go for a Mega

dkmatt0, 2018-06-16:
I optimize the code just a little :
char mqtt_user[**13**] = "homeassistant";
char mqtt_pass[**7**] = "welcome";
#define Gateway_Name "**OMG**"
#define will_Message "**off**"
#define Gateway_AnnouncementMsg "**on**"
#define ota_password **""**

1technophile, 2018-06-28:
Maybe we could close this issue as these 2 gateways simultaneously ate not supported on UNO ?