1technophile/OpenMQTTGateway #215: File *.bin with esp32 ou esp8266

lscosta90br, 2018-06-14:
Hi guys !!

Is there any compiled file for esp32 or esp8266 ... I have difficulty compiling.
thankful
Leandro Costa

Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

So as to have a more efficient treatment of your request please provide the following information:
* OpenMQTTGateway version used (V0.4, 0.3, master...)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries
* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the tchat room below
[image]

lscosta90br, 2018-06-15:
Thank you !!!
I just found

https://github.com/1technophile/OpenMQTTGateway/releases

1technophile, 2018-06-18:
hello, if you are ok with compiling could you close the issue?

lscosta90br, 2018-06-18:
Compiling ok

lscosta90br, 2018-06-18:
compiling  ok

closed 2018-06-18 by lscosta90br