1technophile/OpenMQTTGateway #216: could you make an rfbridge direct like espurna rfbridge direct (hardware mod)

hooijschuur, 2018-06-17:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

So as to have a more efficient treatment of your request please provide the following information:
* OpenMQTTGateway version used (V0.4, 0.3, master...)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries
* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the tchat room below
[image]

1technophile, 2018-06-22:
did you try to upload OMG as it is using (uncommenting) ZgatewayRF only and not ZgatewaySRFB
The config_RF.h should be:
```
    #define RF_RECEIVER_PIN 4
    #define RF_EMITTER_PIN 5
```

1technophile, 2019-04-15:
Im' closing this issue, if you have feedback on your test with Sonoff RF Bridge mod don't hesitate to add it here.

closed 2019-04-15 by 1technophile