1technophile/OpenMQTTGateway #217: Battery/Firmware reading for Xiaomi flora & temperature/humidity sensor

mark078, 2018-06-19:
Hi, I've been using the Openmqttgateway with Fhem for a few days now and it works fine. 

For Fhem there is a Perl module which excludes the readings of the Xiaomi BLE sensors. 
Among other things, it is also possible to read the battery and firmware version. 

gatttool -i $hci -b $mac --char-read -a 0x38";   # Firmware and Battery
Source: https://svn.fhem.de/trac/browser/trunk/fhem/FHEM/74_XiaomiFlowerSens.pm?rev=15916

It would be great if you could put that in too Openmqttgateway. 
So that maybe every 24 hours the battery values are read out.

tunip, 2018-07-23:
For Xiaomi temperature/humidity sensor the battery handle is 0x18 and the firmware handle is 0x24.

Source: https://github.com/ratcashdev/mitemp/blob/master/mitemp_bt/mitemp_bt_poller.py

skeptizzle, 2018-08-24:
I would love support for battery level from Mi Flora as well!

mark078, 2018-08-24:
And for the Xiaomi Oclean SE toothbrush it would also be good.  This works with 0x0011 and the output is hex coded.

gatttool -i hci0 -b 70:28:45:5E:xx:xx:xx --char-read --handle=0x0011

There the battery stand can maybe probably only be read out with the APP.

michel72, 2018-09-30:
+1 for the mi flora battery!

rufik, 2019-04-03:
I'm also waiting for battery info :)

1technophile, 2019-07-25:
Unfortunately the method used here cause instability of the gateway (which is compensed in the current code by the hibernation), after some minutes of hour the gateway hang when trying to connect to mi flora, I will continue working on this but is will not be included in v0.9.2 as planned

ColinShorts, 2019-07-25:
I believe if you're prepared to not send json to mqtt, the battery reporting works fine, at least, it does for me.

1technophile, 2019-07-26:
For mi jia it already works indeed, mi jia broadcasts the battery vamue without connection. For mi flora the program proposed use a ble connection to retrieve the battery value and unfortunately this is generating instability.

ManuelS383, 2020-03-22:
Is this still in work?

tparvais, 2020-08-05:
I compiled 0.9.4 released for ESP32 + BT to capture Tempareture/Humidity sensor. I'm surprised to not see the battery level in MQTT ?

is that normal ?

[image]

1technophile, 2020-08-05:
For this model (LYWSDCGQ) it doesn't advertise `batt `at the same time of `tem `and `hum `, wait and you should see it.

1technophile, 2020-08-06:
> I compiled 0.9.4 released for ESP32 + BT to capture Tempareture/Humidity sensor. I'm surprised to not see the battery level in MQTT ?

Well, I was wrong, I confirm that the battery reading for LYWSDCGQ was not working in v0.9.4. I corrected it in the development branch

tparvais, 2020-08-06:
Possible to get the code updates to let me compile 0.9.4 to enable battery measurements ?

1technophile, 2020-08-06:
> Possible to get the code updates to let me compile 0.9.4 to enable battery measurements ?

This [parameter ](https://github.com/1technophile/OpenMQTTGateway/blob/da63fd3606d5e2331f4b60f0e4e30e350aefc1e9/main/config_BT.h#L42)should be 29

tparvais, 2020-08-06:
> 
> 
> > Possible to get the code updates to let me compile 0.9.4 to enable battery measurements ?
> 
> This [parameter ](https://github.com/1technophile/OpenMQTTGateway/blob/da63fd3606d5e2331f4b60f0e4e30e350aefc1e9/main/config_BT.h#L42)should be 29

i recompiled but no change, no battery level measured

1technophile, 2020-08-06:
Strange, I don't have any issue reading mine:
`{"id":"XXXXXXXXF","rssi":-49,"distance":0.156114,"model":"LYWSDCGQ","batt":100}`

tparvais, 2020-08-07:
Ok it works, I had to wait long time

fizzi, 2021-04-22:
Is there any fix/implementation for MiFlora battery level ?

1technophile, 2021-04-24:
Could be something to do for the next version, let's add it

closed 2021-06-25 by 1technophile

Kernald, 2021-07-21:
I think this is not entirely closed - close, but not yet there. The `bat` attribute is indeed visible in `BTtoMQTT`, but not exposed as a sensor yet - it doesn't generate a `${MAC}-MiFlora-bat` in the same way `${MAC}-MiFlora-moi` is generated for the moisture, for example. I believe the battery would need to be added here: https://github.com/1technophile/OpenMQTTGateway/blob/373ec29803cec5267aaa8fbe05629f9b6c256541/main/ZgatewayBT.ino#L281-L293