1technophile/OpenMQTTGateway #218: SONOFF RF Bridge R2 value always zero

nonsintetic, 2018-06-20:
* Hardware used : Sonoff RF Bridge R2 V1.0
* OpenMQTTGateway version 0.8beta
* RCSwitch version 2.6.2
* RF devices tested: [Remote 1](https://www.aliexpress.com/item/1-key-Mini-RF-433mhz-Universal-Remote-Control-Wireless-Controller/32769093848.html?spm=a2g0s.9042311.0.0.27424c4dXB5cWT) , [Door sensor](https://www.aliexpress.com/item/Wireless-Security-Door-Window-Magnet-Sensor-Detector-with-Battery-433mhz-Motion-Sensor-Home-Alarm/32651890512.html?spm=a2g0s.9042311.0.0.27424c4dXB5cWT) and some others.

I managed to upload the firmware without any issues by following the guide. The device outputs messages to the SRFBtoMQTT topics when I press an RF remote, but the value received on home/OpenMQTTGateway/SRFBtoMQTT is always '0'. There are values being produced on /Raw, /Tsyn, /Thigh and /Tlow but they can't be used for anything. Below is a log of pushing the same button on a remote.

The setup I use and remotes tested *all work fine with OpenMQTTGateway on a NodeMCU + STX882 modules*. Same exact settings except `#define ZgatewaySRFB` instead of  `#define ZgatewayRF`.

```
home/OpenMQTTGateway/SRFBtoMQTT/Raw 2F8A019A04A6000000
home/OpenMQTTGateway/SRFBtoMQTT 0
home/OpenMQTTGateway/SRFBtoMQTT/Tsyn 1217
home/OpenMQTTGateway/SRFBtoMQTT/Thigh 410
home/OpenMQTTGateway/SRFBtoMQTT/Tlow 1190
home/OpenMQTTGateway/SRFBtoMQTT/Raw 2F8A019A04A6000000
home/OpenMQTTGateway/SRFBtoMQTT 0
home/OpenMQTTGateway/SRFBtoMQTT/Tsyn 1217
home/OpenMQTTGateway/SRFBtoMQTT/Thigh 410
home/OpenMQTTGateway/SRFBtoMQTT/Tlow 1190
home/OpenMQTTGateway/SRFBtoMQTT/Raw 2F94019A04A6000000
home/OpenMQTTGateway/SRFBtoMQTT 0
home/OpenMQTTGateway/SRFBtoMQTT/Tsyn 1218
home/OpenMQTTGateway/SRFBtoMQTT/Thigh 410
home/OpenMQTTGateway/SRFBtoMQTT/Tlow 1190
home/OpenMQTTGateway/SRFBtoMQTT/Raw 2F94019A04A6000000
home/OpenMQTTGateway/SRFBtoMQTT 0
home/OpenMQTTGateway/SRFBtoMQTT/Tsyn 1218
home/OpenMQTTGateway/SRFBtoMQTT/Thigh 410
home/OpenMQTTGateway/SRFBtoMQTT/Tlow 1190
```

nonsintetic, 2018-06-21:
Here is the same button being pressed on the working hardware:
```
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 394
home/OpenMQTTGateway/433toMQTT 5592328
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 395
home/OpenMQTTGateway/433toMQTT 5592328

```

1technophile, 2018-06-21:
Could you post also both results also for the door sensors?

1technophile, 2018-06-28:
@nonsintetic do you have some news about your issue ?

nonsintetic, 2018-06-28:
Hello, 

Sorry about the delay, I got caught up with work (meh). Here is the output for all the devices I have around. I hooked up both my diy setup and the bridge and recorded it simultaneously from both:
Remote control ([aliexpress](https://www.aliexpress.com/item/3-key-Mini-RF-433mhz-Universal-Remote-Control-Wireless-Controller/32769484587.html?spm=a2g0s.9042311.0.0.27424c4dbHuQSQ)) :
```
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 395
home/OpenMQTTGateway/433toMQTT 5592325
home/OpenMQTTGateway/SRFBtoMQTT/Raw 2FBC019A04B0000000
home/OpenMQTTGateway/SRFBtoMQTT 0
home/OpenMQTTGateway/SRFBtoMQTT/Tsyn 1222
home/OpenMQTTGateway/SRFBtoMQTT/Thigh 410
home/OpenMQTTGateway/SRFBtoMQTT/Tlow 1200
```
Door sensor ([aliexpress](https://www.aliexpress.com/item/Wireless-Security-Door-Window-Magnet-Sensor-Detector-with-Battery-433mhz-Motion-Sensor-Home-Alarm/32651890512.html?spm=a2g0s.9042311.0.0.27424c4dbHuQSQ))
```
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 419
home/OpenMQTTGateway/433toMQTT 14195225
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 419
home/OpenMQTTGateway/433toMQTT 14195225
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 420
home/OpenMQTTGateway/433toMQTT 14195225
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 420
home/OpenMQTTGateway/433toMQTT 14195225
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 420
home/OpenMQTTGateway/433toMQTT 14195225
home/OpenMQTTGateway/SRFBtoMQTT/Raw 361001B80546D89A19
home/OpenMQTTGateway/SRFBtoMQTT 14195225
home/OpenMQTTGateway/SRFBtoMQTT/Tsyn 1384
home/OpenMQTTGateway/SRFBtoMQTT/Thigh 440
home/OpenMQTTGateway/SRFBtoMQTT/Tlow 1350
```
Fire alarm ([Digoo](https://www.aliexpress.com/item/DIGOO-DG-HOSA-Smart-433MHz-Wireless-Smoke-Detector-Fire-Alarm-Sensor-For-Home-Security-Guarding-Alarm/32823020389.html?spm=a2g0s.9042311.0.0.27424c4dbHuQSQ))
```
home/OpenMQTTGateway/SRFBtoMQTT/Raw 2E5E017C047E787FA8
home/OpenMQTTGateway/SRFBtoMQTT 7897000
home/OpenMQTTGateway/SRFBtoMQTT/Tsyn 1187
home/OpenMQTTGateway/SRFBtoMQTT/Thigh 380
home/OpenMQTTGateway/SRFBtoMQTT/Tlow 1150
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 383
home/OpenMQTTGateway/433toMQTT 7897000
home/OpenMQTTGateway/SRFBtoMQTT/Raw 2E72019A0460787FA8
home/OpenMQTTGateway/SRFBtoMQTT 7897000
home/OpenMQTTGateway/SRFBtoMQTT/Tsyn 1189
home/OpenMQTTGateway/SRFBtoMQTT/Thigh 410
home/OpenMQTTGateway/SRFBtoMQTT/Tlow 1120
```

1technophile, 2018-06-28:
No problem, if I understand well it us working for 2 cases / 3

nonsintetic, 2018-06-28:
I think it may be intermittent. Just after doing the test above the remote started working fine, then I reset the bridge and it seems to be outputting 0 now again. 
When I tested the first time the door sensor and the alarm were generating 0 as well, now they are working fine.
Maybe it's a hardware issue.

nonsintetic, 2018-06-28:
After more testing, the bridge seems to be giving me the same code no matter what device I use. The diy seems to be doing fine:
```
remote button 1
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 396
home/OpenMQTTGateway/433toMQTT 5592325
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 396
home/OpenMQTTGateway/433toMQTT 5592325
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 396
home/OpenMQTTGateway/433toMQTT 5592325
home/OpenMQTTGateway/SRFBtoMQTT/Raw 3020019004B04B57A9
home/OpenMQTTGateway/SRFBtoMQTT 4937641
home/OpenMQTTGateway/SRFBtoMQTT/Tsyn 1232
home/OpenMQTTGateway/SRFBtoMQTT/Thigh 400
home/OpenMQTTGateway/SRFBtoMQTT/Tlow 1200
remote button 2
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 396
home/OpenMQTTGateway/433toMQTT 5592331
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 397
home/OpenMQTTGateway/433toMQTT 5592331
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 397
home/OpenMQTTGateway/433toMQTT 5592331
home/OpenMQTTGateway/SRFBtoMQTT/Raw 2FE4019A04B04B57A9
home/OpenMQTTGateway/SRFBtoMQTT 4937641
home/OpenMQTTGateway/SRFBtoMQTT/Tsyn 1226
home/OpenMQTTGateway/SRFBtoMQTT/Thigh 410
home/OpenMQTTGateway/SRFBtoMQTT/Tlow 1200
door
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 403
home/OpenMQTTGateway/433toMQTT 4937641
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 403
home/OpenMQTTGateway/433toMQTT 4937641
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 403
home/OpenMQTTGateway/433toMQTT 4937641
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 403
home/OpenMQTTGateway/433toMQTT 4937641
home/OpenMQTTGateway/433toMQTT/protocol 1
home/OpenMQTTGateway/433toMQTT/bits 24
home/OpenMQTTGateway/433toMQTT/length 403
home/OpenMQTTGateway/433toMQTT 4937641
home/OpenMQTTGateway/SRFBtoMQTT/Raw 30B6019004BA4B57A9
home/OpenMQTTGateway/SRFBtoMQTT 4937641
home/OpenMQTTGateway/SRFBtoMQTT/Tsyn 1247
home/OpenMQTTGateway/SRFBtoMQTT/Thigh 400
home/OpenMQTTGateway/SRFBtoMQTT/Tlow 1210
```

1technophile, 2018-06-28:
This is strange as you proposed maybe an hardware issue.
If ok for you could you close the issue

closed 2018-06-28 by nonsintetic