1technophile/OpenMQTTGateway #221: Sonoff RF Bridge R2 - blue wifi status led

starkillerOG, 2018-06-27:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

So as to have a more efficient treatment of your request please provide the following information:
* OpenMQTTGateway version used (V0.7)
* Library version related to the problem you have (-)
* For IR and RF clarify if you tested with the basic examples given with these libraries (succesfully received RF signals from PIR, and door sensors)
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

--------------------------------------------

I use the new Sonoff RF Bridge R2 as shown in [this picture](https://user-images.githubusercontent.com/140089/39653002-9d56e7fa-4fef-11e8-9bb3-08e560f28cf6.jpg).
The Green power LED on top works perfectly, the red signal received LED also works perfictly.
However, the blue wifi LED does not do anything, it is always off.

I would love to know the status of the wifi connection from this LED as with the default firmware.
prefered behaviour:
- trying to connect to wifi/broker = blue LED flashing
- connected to wifi/broker = blue LED solid on
- error in connecting to wifi and stopped trying to connect = blue LED off

Is this already possible and do I just need to change some GPIO pin in the configuration somewhere, or is this a new feature request?

1technophile, 2018-06-27:
@8666 implemented[ 3 leds handling](https://github.com/1technophile/OpenMQTTGateway/commit/15cea6b222880cc681bf6d89ac1d8f94232e5bfd#diff-fd1119a40bd6107149eac068bf49f95d) for knowing the gateway state. I don't know if it fits completely to your need but it is a first approach.

You set the pins in[ user_config.h](https://github.com/1technophile/OpenMQTTGateway/commit/15cea6b222880cc681bf6d89ac1d8f94232e5bfd#diff-d526b496c1c8b01772a596ae172bc187)

starkillerOG, 2018-06-27:
Seems like I would need to try out the led_error, that seems to indicate the MQTT connection status.
However I don't know the pin number that the blue led is connected to for the Sonoff RF Bridge R2.

Does anyone know which pin number schould be set for the blue led (indicated by Wifi_LED on the board) on the Sonoff RF Bridge R2?

starkillerOG, 2018-06-28:
In [this issue](https://github.com/xoseperez/espurna/issues/579) on the sepurna github, they clearly discribe that they use the blue wifi led in their code to indicate if the Sonoff RF Bridge is connected to wifi:

> After ~35s the wifi led stops blinking and change to a constant light blue 

So I started searching their code for a LED GPIO pin number for the Sonoff RF Bridge.
I found this in [the hardware.h file](https://github.com/xoseperez/espurna/blob/3c96668df95fece14b12ed26af32cd63dc6881ed/code/espurna/config/hardware.h) of the espurna
 github code:
```

#elif defined(ITEAD_SONOFF_RFBRIDGE)

    // Info
    #define MANUFACTURER        "ITEAD"
    #define DEVICE              "SONOFF_RFBRIDGE"
    #define RELAY_PROVIDER      RELAY_PROVIDER_RFBRIDGE

    // Number of virtual switches
    #ifndef DUMMY_RELAY_COUNT
    #define DUMMY_RELAY_COUNT   8
    #endif

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    1

    // RFB Direct hack thanks to @wildwiz
    // https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack
    #ifndef RFB_DIRECT
    #define RFB_DIRECT          0
    #endif

    #ifndef RFB_RX_PIN
    #define RFB_RX_PIN          4   // GPIO for RX when RFB_DIRECT
    #endif

    #ifndef RFB_TX_PIN
    #define RFB_TX_PIN          5   // GPIO for TX when RFB_DIRECT
    #endif

    // When using un-modified harware, ESPurna communicates with the secondary
    // MCU EFM8BB1 via UART at 19200 bps so we need to change the speed of
    // the port and remove UART noise on serial line
    #if not RFB_DIRECT
    #define SERIAL_BAUDRATE         19200
    #define DEBUG_SERIAL_SUPPORT    0
#endif
```

seems to mee like GPIO 13 could be the blue wifi LED.
Will try it out tonight.

starkillerOG, 2018-06-30:
I tryed pins 13 and 12 but I could not get it to work.
If someone knows which pins to use or how to get it working I would be verry intrested.

Mynasru, 2018-11-22:
I traced back the LEDs to the pins on the ESP8285 & EFM8BB1:
WIFI_LED = GPIO13 (ESP8285)
RF_LED = P0.7 (EFM8BB1)
Big green LED (V1.0 R2) = 3v3

I didn't test it in the firmware yet.

neil2048, 2019-02-16:
Did any one test this yet? and get it working?

closed 2019-08-14 by stale[bot]

m4rcu5, 2019-12-30:
Hi,

I'd like to reopen this issue instead of starting a new one.
By now we are on commit 86f3286, which is the one I am using for this report.

The `WIFI_LED` on the `433RFBRDIGE R2` is indeed connected to `GPIO13`. Defining `led_info 13` will control the LED.
However it seems to be inverted. Lighting most of the time and turning off during transmission.

Would it be possible to implement a inverse option on the 3 leds, like the `#define LED1_PIN_INVERSE` in the Espurna code? This would fully resolve this issue.