1technophile/OpenMQTTGateway #224: time_avoid_duplicate not working

PaulHam211, 2018-07-02:
Wemos D1 with RF STX882 & SRX882

I have upgraded to the 0.7v from a pervious non OpenMQTTGateway version (called 433toMQTTto433).

The pervious version handles duplicate RF codes very well, but it seems to not work at all in the new version. I have tried to change the time_avoid_duplicate value but it doesn't seem to make any difference?

Any ideas?

1technophile, 2018-07-02:
yes, it's a bug in the 0.7[ corrected in 0.8beta](https://github.com/1technophile/OpenMQTTGateway/commit/eb15898228cefec4139f8b359a3ef0a908f9147c)

1technophile, 2018-07-11:
@PaulHam211 I'm closing this issue, if you have other questions don't hesitate to go to the forum or reopen it

closed 2018-07-11 by 1technophile