1technophile/OpenMQTTGateway #227: Generic "Weather Stations" etc

steadramon, 2018-07-05:
Hi! First of all, love the project. I'm putting together a small piece of hardware for my home automation and the project ticks all the boxes expect one - 433mhz weather stations. I have these 'lil 433 temperature and humidity sensors around the house and rather than re-investing in these devices would like to get them working with OMQTTG

From looking into things it seems that RemoteSensor from the fuzzillogic 433mhzforarduino libraries would work - InterruptChain is able to allow both libraries to be used in callbacks - I wonder if this is something you think would be worth looking into for other users? I'm a bit of an Arduino newbie so my pull request may be long in the coming but would appreciate any pointers you might have for me.

1technophile, 2018-07-05:
Hello!
If you could try remotesensor from fuzzylogic on some hardware, arduino first and later ESP and if it works for you, I can try to integrate it. 
So as to integrate new devices efficiently the best way is to have one on hand but I don't have weather station compatible with fuzzylogic library.

steadramon, 2018-07-05:
Hmmm, after testing it doesn't seem to pick up any my remotes, pirs, or the weather stations when using any of the fuzzilogic libs.. RF1 at least gets me those. 
The ESPiLight libs seem to work, but don't pick up my remotes or pirs but they do work with the stations.

So kinda seem a bit stuck, I've had a little go with the ESPiLight libs with OMQTTG and they seem to be able to play nice - I just need a solution that works with both my remotes and my sensors, which it looks like this won't without some significant tweaking of the libraries.

steadramon, 2018-07-11:
The main issue seems to be that I only get responses from the ESPiLight libraries if I load both. 
I've made a ZgatewayPilight, using the same format as the RF2.
I get messages fine from the gateway for the Pilight lib but then nothing from RF. I guess we'd need a standard collector which then feeds the outputs to both pilight and rc-switch?

1technophile, 2018-07-11:
Maybe some conflict with the callbacks.
In all the case it would be interesting that you submit a pull request with ZgatewayPilight.
I think a lot off people could be interested by this. 
And it can be a first base to make both work together.

1technophile, 2018-07-28:
@steadramon would it be possible to share your code so as to enable others to test it and maybe find a way to solve your issue?

1technophile, 2018-09-07:
I'm closing the issue as there is a PR in reviewing process

closed 2018-09-07 by 1technophile