1technophile/OpenMQTTGateway #231: Add YS-IRTM (all in one IR send&receive module) support [enhancement] [help wanted]

rradar, 2018-07-11:
Hello,

we would like to see support for the YS-IRTM (all in one IR send&receive module)

[image]

@prahjister forked a old version here with support for the device here: https://github.com/prahjister/433toMQTTto433_ESP8266

rradar, 2018-07-12:
There is more information here: https://www.cnx-software.com/2017/04/20/karls-home-automation-project-part-4-mqtt-bridge-updated-to-use-ys-irtm-ir-receiver-transmitter-with-nodemcu/

The code for the hardware should be this one:

```
…….

// software serial : TX = digital pin 12, RX = digital pin 14
SoftwareSerial portOne(14, 12);
………….

#define subject433toMQTT "home/433toMQTT"
#define subjectMQTTto433 "home/commands/MQTTto433"

……….
  
  // Start IR software serial ports
  portOne.begin(9600);
  
…………
  
//begining of IR
if (portOne.available() > 0) {
delay(1);
int my_in_bytes[3]={0, 0, 0};

  for (int i=0; i <= 2; i++){
    incomingByte = portOne.read();
    Serial.print("I received: ");
    Serial.println (String (incomingByte, HEX));
    my_in_bytes [i] = incomingByte;
  }
String IRvalue;
  for (int i=0; i <= 2; i++){
    Serial.print(String (my_in_bytes [i], HEX) + "," );
    IRvalue = IRvalue + (String (my_in_bytes [i], HEX) + "," );
  }
    Serial.println("");
    sendMQTT(subjectIRtoMQTT,String(IRvalue));
  }
 
…………….

//IR Recieved
  if (topicNameRec = subjectMQTTtoIR){
  char input1[3];
  char input2[3];
  char input3[3];
  int val1;
  int val2;
  int val3;
   
    uint8_t my_out_bytes[5]={0xA1, 0xF1, 0, 0, 0};

    int commaIndex = callbackstring.indexOf(',');
    //  Search for the next comma just after the first
    int secondCommaIndex = callbackstring.indexOf(',', commaIndex + 1);
    
    String firstValue = callbackstring.substring(0, commaIndex);
    String secondValue = callbackstring.substring(commaIndex + 1, secondCommaIndex);
    String thirdValue = callbackstring.substring(secondCommaIndex + 1); // To the end of the string
    firstValue.toCharArray(input1, 4);
    val1 = StrToHex(input1);
    my_out_bytes [2] = val1;
    secondValue.toCharArray(input2, 4);
    val2 = StrToHex(input2);
    my_out_bytes [3] = val2;
    thirdValue.toCharArray(input3, 4);
    val3 = StrToHex(input3);
    my_out_bytes [4] = val3;

    for (int i=0; i <= 4; i++){
    Serial.print(String (my_out_bytes [i], HEX) + "," );
    }
    portOne.write(my_out_bytes,sizeof(my_out_bytes));
    Serial.println("");

  }

……….
//IR String to Hex Conversion

int StrToHex(char str[])
{
  return (int) strtol(str, 0, 16);
}
```

sanderpleijers, 2018-12-04:
I have those modules, so if you need testers?

rradar, 2018-12-06:
@sanderpleijers 
I'm still struggling where to put the code. I would love to see this module supported. I also could test a sketch if possible.

Here is a link (https://www.cnx-software.com/2017/04/20/karls-home-automation-project-part-4-mqtt-bridge-updated-to-use-ys-irtm-ir-receiver-transmitter-with-nodemcu/) which is documented quite well but I would love to have it upstream supported by OpenMQTTGateway to be able to update without have to hassle for the custom made changes again and again.

Anyway I would be happy if anyone can offer a working sketch :+1: :1st_place_medal:

1technophile, 2023-02-04:
Closing it, PR welcome

closed 2023-02-04 by 1technophile